openthread/openthread #7811: Delayed network data update

abtink, 2022-06-14:
We have noticed failures cases of `test-026-slaac-address-wpantund` toranj test-case.
Debugging and checking the log I found this pattern (adding a simplified version of events that can produce the issue):
- Network topology is a leader and a router.
- Leader node registers an on-mesh prefix and network data is updated on all nodes.
- Leader is now rebooted (looses all previous netdata info)
- Leader starts agian, trying to become leader again
   -  Send "Link Request" to all and get a "Link Accept" from router, so becomes leader
   - Send a "Data Request" to get the network data (but router delays sending the response).
- Next layer code (`wpantund`) adds another on-mesh prefix which will be added quickly (since device itself is leader).
- Leader will suppress sending "Data Response" to all to notify of the netdata change (since it is still waiting to retrive netdata itself).
- Then we receive a "Data Response" from the router (which contain all the old entries and the previous prefix added by leader)
- "Data Response" replaces the network data on leader (we loose the newly added entry).
- But this should be temporary since the `NetworkData::Notifier` logic would detect inconsistency and re-register the local network data
- The problem is that this last update can take up to 5 minutes.

In `NetworkData::Notifier` we have code which delays checks after a previous successful registration with leader:
- After we successfully register local netdata with leader, we use `mNextDelay = kDelaySynchronizeServerData` (which is set to  `300000` msec or 300 secs (start a timer with this delay).
- If the netdata gets changed, we do call `SynchronizeServerData()` but it will exit if timer is running. So we can wait up to 5 min.

Some ideas on this:
- We can add logic that if we are leader and get a "Data Response" from a another node (during initial recovery), we inform the `NetworkData::Notifier` to stop the timer (this would specifically address the situation above)
  - But I feel this may still happen if another node regsiters with leader (before it receives the "Data Response"). 
- Other ideas can be to change/relax the delay logic in `NetworkData::Notifier` but it probably require more considerations/thoughts.

jwhui, 2022-06-14:
@abtink , thanks for raising this issue!

Ideally, the Leader should not be accepting any new network data from any Thread devices (including itself) while it is waiting to synchronize Thread network data. One possibility is not to respond to the SVR_DATA.ntf message and let the CoAP transaction time out? With the existing CoAP parameters, I think that is 45 seconds (which is better than the 5 minutes).

Thoughts?

simonlingoogle, 2022-06-15:
Shouldn't we synchronize network data quickly if the timer was scheduled with a long duration?

jwhui, 2022-06-15:
> Shouldn't we synchronize network data quickly if the timer was scheduled with a long duration?

@simonlingoogle , can you elaborate? I'm not sure what you are suggesting.

jwhui, 2022-06-16:
> Seems `SynchronizeServerData` was called but it exit immediately because the Timer was running. I think the logic behind it is that `SynchronizeServerData` needs doing nothing since it will be called again soon. However, we failed to consider the case where the Timer was scheduled with a long duration. In the case of this issue, I think we need to synchronize the network data immediately or reduce the timer delay to be short enough.

A primary reason for the timer is to rate-limit SVR_DATA.ntf messages. I a device should send a SVR_DATA.ntf message any time it notices a Thread Network Data change that does not include its desired information. Ideally, we should be special-casing when a newly restored Leader is trying to resynchronize its Thread Network Data. However, Thread does not provide a real way to respond to a SVR_DATA.ntf indicating that the Leader is not yet ready. However, one way to do this is simply not responding to SVR_DATA.ntf.

One could consider that a newly restored Leader that has not yet synchronized its Thread Network Data is not really a Leader yet. Not responding to SVR_DATA.ntf is equivalent to the Leader not being available.

simonlingoogle, 2022-06-17:
> However, Thread does not provide a real way to respond to a SVR_DATA.ntf indicating that the Leader is not yet ready. However, one way to do this is simply not responding to SVR_DATA.ntf.

What if the Link Request is never successful? When should the Leader stop rejecting SVR_DATA.ntf ?

jwhui, 2022-06-17:
> What if the Link Request is never successful? When should the Leader stop rejecting SVR_DATA.ntf ?

Link Request has to succeed before the device even becomes a Leader.

https://github.com/openthread/openthread/blob/d5926a2445ba30200ec7a41c2ccc163cc9bf5b71/src/core/thread/mle_router.cpp#L939

abtink, 2022-06-17:
> Ideally, the Leader should not be accepting any new network data from any Thread devices (including itself) while it is waiting to synchronize Thread network data. One possibility is not to respond [...]

Thanks guys. I agree. @jwhui  suggestion above sounds good to me. I was thinking something similar. 
- Basically leader after recovery should not accept new netdata registrations while still waiting to retrieve net data itself.
- I think we may also need  a mechanism for (I guess rare but possible) situation if leader does not get net data (e.g. the neighboring router gets reset after sending "Link Accept" but before it can respond with a "Data Response"). Maybe a timeout after which leader would form a new partition (which I guess help all other nodes re-register as well).

I will try to add a PR with this idea later.

closed 2022-06-30 by jwhui