openthread/openthread #7835: The thread sub-device failed to register the srp service with otbr

luyou-tuya, 2022-06-24:
**Describe the bug** A clear and concise description of what the bug is.
After the thread joins the thread network, when registering the srp service with otbr, the otbr log shows the following error，what is the possible reason:
**To Reproduce** Information to reproduce the behavior, including:
otbr-agent[3322]: 00:03:24.703 [I] MeshForwarder-:     dst:[fd05:c104:ef7c:a2a3:28fe:eae9:2dc1:9ce7]:53535
otbr-agent[3322]: 00:03:24.703 [I] SrpServer-----: Received DNS update from fd05:c104:ef7c:a2a3:300c:f440:f89:cd4d
otbr-agent[3322]: 00:03:24.703 [W] SrpServer-----: Failed to process DNS Additional section: Failed
otbr-agent[3322]: 00:03:24.704 [W] SrpServer-----: Send fail response: 5
otbr-agent[3322]: 00:03:24.724 [I] MeshForwarder-: Sent IPv6 UDP msg, len:60, chksum:75bb, ecn:no, to:0xc800, sec:yes, prio:low
otbr-agent[3322]: 00:03:24.724 [I] MeshForwarder-:     src:[fd05:c104:ef7c:a2a3:28fe:eae9:2dc1:9ce7]:53535
otbr-agent[3322]: 00:03:24.724 [I] MeshForwarder-:     dst:[fd05:c104:ef7c:a2a3:300c:f440:f89:cd4d]:49154
otbr-agent[3322]: 00:03:25.624 [I] Mac-----------: Frame rx failed, error:UnknownNeighbor, len:28, seqnum:101, type:Beacon, src:0x0000, dst:None, sec:no, ackreq:no
thread otbr git commit 3a98779dc
1. Git commit id
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

wgtdkp, 2022-06-24:
Please upload full OTBR logs, and could you provide more information about the SRP client device?
1. commit ID
2. What services are being registered (`srp client service`)

luyou-tuya, 2022-06-24:
> Please upload full OTBR logs, and could you provide more information about the SRP client device?
> 
> 1. commit ID
> 2. What services are being registered (`srp client service`)

the otbr commit Id 3a98779dc
thread sub-device in matter Commissioning
  During the process, the registration of the srp service failed after joining the otbr thread network, the thread sub-device log is gone, and the client log will be provided next time.

jwhui, 2022-06-24:
@luyou-tuya, if you could help provide a sniffer packet capture, that would also help.

luyou-tuya, 2022-06-24:
> @luyou-tuya, if you could help provide a sniffer packet capture, that would also help.

ok！

wgtdkp, 2022-06-24:
I suspect that this is IPv6 MTU issue (https://github.com/openthread/openthread/issues/7742) which is fixed by https://github.com/openthread/openthread/pull/7776. Please try syncing otbr code to main HEAD.

closed 2022-06-30 by luyou-tuya

wgtdkp, 2022-06-30:
@luyou-tuya Is the issue resolved (how?) or obsoleted?