openthread/openthread #7852: What is the reason? The thread sub-device fails to register the srp service prompt:OnSrpClientNotification: Timed out waiting on server response

luyou-tuya, 2022-06-30:
**Describe the bug** A clear and concise description of what the bug is.
What is the reason?I don't know the specific reason, maybe it's related to threadI don't know the specific reason. It may be related to thread. What is the process of thread sub-device registering srp service with otbr? Can anyone provide some help?
During the MATTER Commissioning process of the thread sub-device, after the ThreadNetworkEnable step is executed, the thread sub-device always fails to register the srp service with otbr, and the thread sub-device log prompts OnSrpClientNotification: Timed out waiting on server response, execute ./ot-ctl on the otb side The srp server service command cannot see the corresponding srp service of the sub-device
      Looking at the log of the sub-device, the sub-device has been successfully added to the thread, the status has become child, and the corresponding IP address has been obtained. The sub-device log is as follows:
SRP Client was started, detected server:
fdbb:938a:8493:f80c:0e94:8902:29e3:3032
00> <tuya matter info > [DL] [timeMs 66282 ]_OnPlatformEvent default: event->Type = 32780
00> <tuya matter detail> [DL] [timeMs 66282 ]OpenThread State Changed (Flags: 0x200012a4)
00> <tuya matter detail> [DL] [timeMs 66282 ] Device Role: CHILD
00> <tuya matter detail> [DL] [timeMs 66282 ] Partition Id: 0x7D70AF2C
00> <tuya matter detail> [DL] [timeMs 66282 ]Thread Attached updating Multicast address
00> <tuya matter info > [SVR] [timeMs 66282 ]Joining Multicast groups
00> <tuya matter log >[timeMs 66284 ]### kThreadStateChange ###
00> <tuya matter log >[timeMs 66284 ]# RoleChanged: 1
00> <tuya matter log >[timeMs 66284 ]# AddressChanged: 0
00> <tuya matter log >[timeMs 66284 ]# NetDataChanged: 1
00> <tuya matter log >[timeMs 66284 ]# ChildNodesChanged: 0
00> <tuya matter log >[timeMs 66284 ]# AddressChanged: 0x200012a4
00> <tuya matter info > [DL] [timeMs 66286 ]_OnPlatformEvent default: event->Type = 32780
00> <tuya matter detail> [DL] [timeMs 66286 ]OpenThread State Changed (Flags: 0x00000001)
00> <tuya matter detail> [DL] [timeMs 66288 ] Thread Unicast Addresses:
00> <tuya matter detail> [DL] [timeMs 66288 ] fd92:360:e81a:1:261c:352e:c0c0:a6bd/64 valid preferred
00> <tuya matter detail> [DL] [timeMs 66288 ] fdbb:938a:8493:f80c:0:ff:fe00:3/64 valid rloc
00> <tuya matter detail> [DL] [timeMs 66290 ] fdbb:938a:8493:f80c:416e:d7e4:4207:b7e1/64 valid
00> <tuya matter detail> [DL] [timeMs 66290 ] fe80:0:0:0:d427:4b74:b5ad:b6d0/64 valid preferred
00> <tuya matter detail> [DL] [timeMs 66291 ]Thread Attached updating Multicast address
00> <tuya matter info > [SVR] [timeMs 66291 ]Joining Multicast groups
Write request/command received for CHIPoBLE RX characteristic (con 1, len 3)
00> <tuya matter info > [DL] [timeMs 67853 ]_OnPlatformEvent kCHIPoBLEWriteReceived
00> <tuya matter error > [DL] [timeMs 69362 ]OnSrpClientNotification: Timed out waiting on server response
00> <tuya matter info> [DL] [timeMs 70697 ]_OnPlatformEvent kCHIPoBLENotifyConfirm

The log on the app side shows:
Operational discovery failed for 0x00000000000000ED: ../../../../../../../../../../../osbase-matter/src/lib/address_resolve/AddressResolve_DefaultImpl .cpp:174: CHIP Error 0x00000032: Timeout
**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id
otbr commit id 3a98779dc
3. IEEE 802.15.4 hardware platform
4. Build steps
5. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

wgtdkp, 2022-06-30:
Hi, could you provide the **OTBR full log**?

luyou-tuya, 2022-06-30:
> Hi, could you provide the **OTBR full log**?
Feel sorry! Every time a colleague forgets to grab the otb log, it happens occasionally, and it always appears when it appears

luyou-tuya, 2022-07-19:
@wgtdkp Hi! I got the otbr log. When the sub-device joins the thread network for the first time, the otbr log shows "Child timeout expired", and the log information is otbr1.txt, subdevice1.txt.
The second log thread sub-device joins the thread network, the sub-device displays "timed out waiting on server response", log information otbr1.txt, subdevice2.txt
Please help analyze the log information, the log will be viewed from the end, thank you！
[subdevice1.txt](https://github.com/openthread/openthread/files/9141687/subdevice1.txt)

[subdevice2.txt](https://github.com/openthread/openthread/files/9141690/subdevice2.txt)

[otbr1.txt](https://github.com/openthread/openthread/files/9141688/otbr1.txt)

[otbr2.txt](https://github.com/openthread/openthread/files/9141697/otbr2.txt)

jwhui, 2022-09-23:
Closing stale issue.

closed 2022-09-23 by jwhui