openthread/openthread #7899: [mac] mac counters are 'incorrect' in RCP mode

Irving-cl, 2022-07-14:
**Describe the bug** A clear and concise description of what the bug is.
Not sure if this is a bug. I think it is.

In RCP mode, when a re-tx happens on RCP side (for example, because of channel access failure), in `SubMac::HandleTransmitDone`:
```
    shouldRetx = ((aError != kErrorNone) && ShouldHandleRetries() && (mTransmitRetries < aFrame.GetMaxFrameRetries()));

    mCallbacks.RecordFrameTransmitStatus(aFrame, aAckFrame, aError, mTransmitRetries, shouldRetx);

    if (shouldRetx)
    {
        mTransmitRetries++;
        aFrame.SetIsARetransmission(true);

#if OPENTHREAD_CONFIG_MAC_ADD_DELAY_ON_NO_ACK_ERROR_BEFORE_RETRY
        if (aError == kErrorNoAck)
        {
            SetState(kStateDelayBeforeRetx);
            StartTimerForBackoff(mRetxDelayBackOffExponent);
            mRetxDelayBackOffExponent = OT_MIN(mRetxDelayBackOffExponent + 1, kRetxDelayMaxBackoffExponent);
            ExitNow();
        }
#endif

        StartCsmaBackoff();
        ExitNow();
    }

    SetState(kStateReceive);

    mCallbacks.TransmitDone(aFrame, aAckFrame, aError);

```
RCP will do the re-tx. Only when the tx succeeds or failure count exceeds the limit, `mCallbacks.TransmitDone` will be called, the `LinkRaw` callback will be called and the 'error' will be passed to the 'Mac' layer on host. That means, counters like retries and cca Errors will not be seen by mac counters. For example, when the RCP does the transmission, it retries 2 times, but that won't be reflected in mac counters. That makes it hard to do performance analysis.

**Proposal**
I think we can handle the counters handling for RCP in `RecordFrameTransmitStatus`. And when querying mac counters, we will still need to query the RCP to get the correct counters. That will make the code a little more complex. This is the best way I can think of. What do you think? @abtink @bukepo @zhanglongxia

abtink, 2022-07-14:
@Irving-cl this is not a bug.

The MAC counters related to retry is applicable when OT `SubMac` is in charge of performing frame retx.  If the radio platform takes care of frame retx (i.e. it provides `OT_RADIO_CAPS_TRANSMIT_RETRIES` capability), these counter are not meaningful anymore (the radio platform will do retx and report the final status in `TxDone` callback to OT).

Note that RCP architecture falls under this situation, i.e. from the perspective of OT code running on host, RCP is a radio platform implementation which provide `OT_RADIO_CAPS_TRANSMIT_RETRIES`. 

My suggestion (considering if we want to enhance this) is to have a solution which works in general case when radio platform handles frame retx (instead of RCP-specific solution). 
- We can consider updating the radio platform APIs to add a mechanism (probably under a radio capbility)  for the radio platform to report back number of tx attempts  when it report tx-done. 
- For example, we can add a field in `otRadioFrame` like `mTxAttempts` which we expect to be populated by radio platform  on frame passed in `otPlatRadioTxDone()`.

jwhui, 2022-08-23:
Closing stale issue.

closed 2022-08-23 by jwhui