openthread/openthread #7907: [csl] parent changes (while attached) may not update the CSL parent info on radio

abtink, 2022-07-16:
This is follow up from https://github.com/openthread/openthread/pull/7874 and https://github.com/openthread/openthread/pull/7783.

- On a child node, the parent can change while device is staying attached (e.g. from periodic parent search), 
- On a change on MLE we do call `SetStateChild()` which will trigger a call to `Mac::UpdateCsl()`
- This will trigger a call to `SubMac::UpdateCsl()` with new parent info (parent short and extended addresses) from 
```c++
      if (mLinks.UpdateCsl(period, channel, Get<Mle::Mle>().GetParent().GetRloc16(),
                         &Get<Mle::Mle>().GetParent().GetExtAddress()))
```
- But then `SubMac::UpdateCsl()` may return early before updating the info on radio (since neither period nor channel are changed).
```c++
bool SubMac::UpdateCsl(uint16_t aPeriod, uint8_t aChannel, otShortAddress aShortAddr, const otExtAddress *aExtAddr)
{
    bool diffPeriod  = aPeriod != mCslPeriod;
    bool diffChannel = aChannel != mCslChannel;
    bool retval      = diffPeriod || diffChannel;

    VerifyOrExit(retval);
    mCslChannel = aChannel;

    VerifyOrExit(diffPeriod);
    mCslPeriod = aPeriod;
    IgnoreError(Get<Radio>().EnableCsl(aPeriod, aShortAddr, aExtAddr));
```

I think this may be an issue. Documenting this here so we can fix it later.

Just FYI: @edmont @Irving-cl

edmont, 2022-07-18:
I'm trying to address this issue here, please check: https://github.com/openthread/openthread/pull/7911

abtink, 2022-07-18:
Thanks @edmont (you beat me to adding the change 😃).
I agree saving the peer short address on `SubMac` make sense.

closed 2022-07-18 by jwhui