openthread/openthread #7923: Joining process takes very long or never happens

mwilniewiec, 2022-07-20:
We are using openthread in a busy ether where we have different networks & lots of nodes.
In such environment it takes forever for a node to join.

**To Reproduce**
Prepare a busy environment with multiple networks & many nodes.
Try to join a node to a specific network.

**Expected behavior** 
Node join process should finish with "Join Success" as one commissioner is ale to let this node in.

**Current behavior**
Node tries to join multiple times to the network we are not expecting and fails with "Join Failed".
Multiple tries of joining finishes with the same result.

**What happens**
First of all `OPENTHREAD_CONFIG_JOINER_MAX_CANDIDATES = 2` by default.
This means that node will try to join only to "best" 2 nodes from discovery result.
If all "best" nodes are nodes from a network that we are not expecting - "Join success" will never happen.
How about additional PANID checking so that we can have different PANIDs in the mJoinerRouters table?
I would say that having 1 "best" node from every discovered PANID should be enough.

jwhui, 2022-07-20:
@mwilniewiec , thanks for raising this issue.

Thread's commissioning process assumes that networks will be configured with the Joiner ID(s) that they are looking to join. These Joiner IDs are encoded in Steering Data TLVs included in the Discovery Response messages. The Joiner looks for its Joiner ID in the received Steering Data TLVs to filter the set of networks it will attempt to join. In this scenario, we expect the number of networks (i.e. PAN IDs) to be small (typically one) and we want the device to try different neighbors if attempts are failing.

Are you able to use the existing Steering Data TLV mechanism to help with filtering?

mwilniewiec, 2022-07-25:
@jwhui thanks for the answer. 
I was aware that there is something like Steering Data and Bloom Filters, but I was pretty sure we checked it long time ago and for some reason we omitted it.
From what I see it may be useful and we can try it but it's more complex to us as our commissioner is currently not aware of joiners that are expected to join. Join attempts are transferred directly to cloud and commissioner receives a a joiner PSKd or a reject.

We first need to make some joiner testing with extra OT logs as we see that our `OPENTHREAD_CONFIG_JOINER_MAX_CANDIDATES` is currently set to `65` which is really high but joining process doesn't match our understanding and we don't know what we are missing.

I think we can also try an OT patch that will put more PANIDs to a `mJoinerRouters` table as its really simple to write.

Later we will try to rearrange our joining process to introduce steering data on the commissioner.
One thing that be problematic is our network sizes (up to 100 devices). In such cases collisions in bloom filters are really high:
```
n=30 p=0.140
n=50 p=0.294
n=100 p=0.625
```
Still... We have lots of networks around 30 devices so we can benefit there.

It looked wrong to me that with default settings joiner may never join to an expected network.
But as you said... Using steering data + increasing `OPENTHREAD_CONFIG_JOINER_MAX_CANDIDATES` should potentially fix the problem.

Closing for now, thanks for help

closed 2022-07-25 by mwilniewiec

mwilniewiec, 2022-08-11:
@jwhui 
As I sad earlier we had to investigate problem a little bit more as we had different expectations from the code we read.
Things that we found:
1. In busy environment our Commissioner (NCP) is not found in a discover result, it is almost always found in `scan` result, but hardly ever in `discover`. Why the difference between the two is so huge? PS. changing commissioner channel helps as expected.
2. We see that CoapSecure.Connect() gives us callback after 2 minutes (when joiner is rejected), possibly we close DTLS session in a wrong way on the commissioner side. What is a correct way to do that?
3. We see that there are lots of duplicates (pairs of extAddresses) in discover/scan result, is it expected? Shouldn't we have only unique entries in `mJoinerRouters` in `joiner.cpp` ?