openthread/openthread #7931: Error with using external heap and having message buffers use the heap allocator

silabs-nino, 2022-07-21:
Observing compilation errors when enabling `OPENTHREAD_CONFIG_MESSAGE_USE_HEAP_ENABLE`. Specifically, the error is due to 

> `'GetHeap' is not a member of 'ot::Instance'	message.cpp`

On first inspection this appears to be due to the usage of the external heap configuration (`OPENTHREAD_CONFIG_HEAP_EXTERNAL_ENABLE`). Un-enabling this feature causes the compilation errors to go away.

Looking at the source, if OPENTHREAD_CONFIG_MESSAGE_USE_HEAP_ENABLE is enabled then MessagePool::GetFreeBufferCount in  message.cpp will use GetHeap() to determine the number of buffers available.
```
uint16_t MessagePool::GetFreeBufferCount(void) const
{
    uint16_t rval;

#if OPENTHREAD_CONFIG_MESSAGE_USE_HEAP_ENABLE
    rval = static_cast<uint16_t>(Instance::GetHeap().GetFreeSize() / sizeof(Buffer));
#elif OPENTHREAD_CONFIG_PLATFORM_MESSAGE_MANAGEMENT
    rval = otPlatMessagePoolNumFreeBuffers(&GetInstance());
#else
    rval = mNumFreeBuffers;
#endif

    return rval;
}
```
However, it looks like the GetHeap method is not declared if OPENTHREAD_CONFIG_HEAP_EXTERNAL_ENABLE  is true (instance.hpp).
```
#if !OPENTHREAD_CONFIG_HEAP_EXTERNAL_ENABLE
    /**
     * This method returns a reference to the Heap object.
     *
     * @returns A reference to the Heap object.
     *
     */
    static Utils::Heap &GetHeap(void) { return sHeap; }
#endif
```

The expectation would be that it is possible to use external heap and have message buffers use the heap allocator.

abtink, 2022-07-21:
Thanks @silabs-nino . I expect it should possible to use such combination of configs (exteranl heap along with msg buffer using heap).

The `GetHeap()` seems to be used in `Message` methods which get an estimate of total/remaining buffers (which are used when getting buffer info ). We can consider adding checks to return zero or some specific "invalid" value indicating no estimate is available when we have such a combination.

abtink, 2022-07-21:
Just submitted PR:
- https://github.com/openthread/openthread/pull/7933

closed 2022-07-22 by jwhui