openthread/openthread #7976: Coap blockwise  issue - token missing

dilipzi, 2022-08-03:
**Describe the bug** A clear and concise description of what the bug is.
Coap clockwise- doesn't work. Issue is all Coap block's header should carry same token number. But in our observation from client side first Coap block was having token number while same token number was not set for 2nd and consecutive token numbers.
Version
#define OT_THREAD_VERSION_INVALID 0
#define OT_THREAD_VERSION_1_1 2
#define OT_THREAD_VERSION_1_2 3

We have used Silicon lab MG24 chip.

TO  reproduce issue
1. Just define block size  -  may be 1024 bytes
2. Try to send 1500 bytes of data. This requires two blocks of data to be sent.
3. We can observe that only 1024 bytes of data received by server, while 2nd block and remaining bytes doesn't reach server.

We added below fix.
[coap.zip](https://github.com/openthread/openthread/files/9250148/coap.zip)
File name:             src\core\coap.cpp
method name:  CoapBase::PrepareNextBlockRequest
    SuccessOrExit(error = iterator.Init(aRequestOld));
++    SuccessOrExit(error = aRequest.SetTokenFromMessage(aRequestOld));

abtink, 2022-08-03:
Thanks @dilipzi  The fix/change seems reasonable to me. 
Please feel free to submit a PR with this change, or if you want I can help submit this (please let me know).

abtink, 2022-08-03:
I went ahead and submitted https://github.com/openthread/openthread/pull/7980. Thanks.

dilipzi, 2022-08-04:
Thanks Abtin,

Regards
Dilip

EXTERNAL EMAIL

jwhui, 2022-08-05:
@dilipzi , thanks for raising this issue.

@dilipzi @abtink , can you help provide a pointer to RFC text that requires the CoAP Token to be the same across different GET requests for different blocks? As far as I know, the ETag is intended to be used for correlating blocks of a given representation - see [RFC 7959 Section 2.4](https://www.rfc-editor.org/rfc/rfc7959.html#section-2.4).

midcontinentcontrols, 2022-08-17:
I have also encountered this issue. I can't seem to find a definitive answer on the use of tokens for block correlation, though it seems that they are not intended to be used as such per [RFC7959 - 3.4](https://www.rfc-editor.org/rfc/rfc7959#section-3.4) (see paragraph under figure 12). 

I have been working through a few changes to get enough functionality for our FTD to communicate with a linux host running libcoap. Most block transfers do not work in this case, mostly due to the options not being appropriately copied in certain circumstances. 

libcoap generates stateless tokens per each block past the original, such that each block in the transfer has a different token. I ended up generating random tokens for each block, though I don't think it is necessary.

As is, it is possible for options to be copied out of order especially when user defined options with number higher than block1/2 are present. Here's what I've done wrt option processing. I have tested this only between a nrf52840dk FTD <-> linux host running libcoap, all of block2 tx/rx and block1 tx/rx work with these changes. Previously only block1 rx worked. This has NOT been tested for simultaneous use of block1/2 in a transfer.

```diff
@@ -577,7 +585,6 @@ Error CoapBase::PrepareNextBlockRequest(Message::BlockType aType,
                                         Message &          aMessage)
 {
     Error            error       = kErrorNone;
-    bool             isOptionSet = false;
     uint64_t         optionBuf   = 0;
     uint16_t         blockOption = 0;
     Option::Iterator iterator;
@@ -585,55 +592,42 @@ Error CoapBase::PrepareNextBlockRequest(Message::BlockType aType,
     blockOption = (aType == Message::kBlockType1) ? kOptionBlock1 : kOptionBlock2;
 
     aRequest.Init(kTypeConfirmable, static_cast<ot::Coap::Code>(aRequestOld.GetCode()));
+    IgnoreReturnValue(aRequest.GenerateRandomToken(aRequestOld.GetTokenLength()));    
+    
+    /// NOTE: Options MUST be added in ascending order
     SuccessOrExit(error = iterator.Init(aRequestOld));
-
-    // Copy options from last response to next message
-    for (; !iterator.IsDone() && iterator.GetOption()->GetLength() != 0; error = iterator.Advance())
+    while (!iterator.IsDone())
     {
-        uint16_t optionNumber = iterator.GetOption()->GetNumber();
-
-        SuccessOrExit(error);
-
-        // Check if option to copy next is higher than or equal to Block1 option
-        if (optionNumber >= blockOption && !isOptionSet)
+        if(iterator.GetOption()->GetLength() != 0) 
         {
-            // Write Block1 option to next message
-            SuccessOrExit(error = aRequest.AppendBlockOption(aType, aMessage.GetBlockWiseBlockNumber() + 1, aMoreBlocks,
-                                                             aMessage.GetBlockWiseBlockSize()));
-            aRequest.SetBlockWiseBlockNumber(aMessage.GetBlockWiseBlockNumber() + 1);
-            aRequest.SetBlockWiseBlockSize(aMessage.GetBlockWiseBlockSize());
-            aRequest.SetMoreBlocksFlag(aMoreBlocks);
-
-            isOptionSet = true;
+            uint16_t optionNumber = iterator.GetOption()->GetNumber();
 
-            // If option to copy next is Block1 or Block2 option, option is not copied
-            if (optionNumber == kOptionBlock1 || optionNumber == kOptionBlock2)
+            if (optionNumber == blockOption)
             {
-                continue;
+                // Replace the block option with params for the next block
+                SuccessOrExit(error = aRequest.AppendBlockOption(aType, aMessage.GetBlockWiseBlockNumber() + 1, aMoreBlocks,
+                                                                aMessage.GetBlockWiseBlockSize()));
+                aRequest.SetBlockWiseBlockNumber(aMessage.GetBlockWiseBlockNumber() + 1);
+                aRequest.SetBlockWiseBlockSize(aMessage.GetBlockWiseBlockSize());
+                aRequest.SetMoreBlocksFlag(aMoreBlocks);
+            }
+            else 
+            {
+                // Copy option
+                SuccessOrExit(error = iterator.ReadOptionValue(&optionBuf));
+                SuccessOrExit(error = aRequest.AppendOption(optionNumber, iterator.GetOption()->GetLength(), &optionBuf));
             }
         }
 
-        // Copy option
-        SuccessOrExit(error = iterator.ReadOptionValue(&optionBuf));
[... 140 lines cut ...]
        {
            hmac.Update((const void*)&optionNumber, sizeof(optionNumber));

            if (optionLength != 0) 
            {
                uint8_t buf[optionLength];
                SuccessOrExit(iterator.ReadOptionValue((void*)buf));
                hmac.Update((const void*)buf, sizeof(optionLength));
            }
        }
        
        IgnoreReturnValue(iterator.Advance());
    }

exit:
    if(error == kErrorNone) {
        hmac.Finish(*static_cast<Crypto::HmacSha256::Hash*>(aHashOut));
        return error;
    }

    return kErrorParse;
}
```

I then call the new otCoapDeriveCacheKey before each blockwise RX/TX hook call and pass the cache key up to the user so the cache can be implemented there. I'm not sure if this is the best way to do it but it does work for me. With this approach, otCoapDeriveCacheKey should be added to the api as well to allow the user to generate cache keys for their TX blockwise hook setup.

EskoDijk, 2022-09-30:
Just adding a note to confirm that Token is not used by CoAP blockwise transfers for indication of a "session" of multiple block transfers. Rather, it is just used as defined by RFC 7252 so a client may use a token or may use the default token (0 bytes token) in case it doesn't need the token value in the response. The token can be selected individually for each single transaction (CoAP req). It is used to match a received response to the proper (single) request.

The real issue flagged here seems to be that Block1 blockwise transfer isn't working. If that works or not should not depend on the Token values used by the client.

If a packet capture could be attached, that would be helpful in the analysis.

closed 2022-12-07 by jwhui