openthread/openthread #8009: [routing-manager] treatment of domain prefixes

jwhui, 2022-08-10:
I see some issues with the current code (because the code in 'main' already changed I think since the merge of this PR, I used the 'main' code branch as a reference for routing-manager).  Basically the question of whether a prefix is a valid OMR prefix is separate from whether the prefix should be advertised in a RIO. Per the spec 9.6.1.2, a Domain Prefix (P_dp=1) is a valid OMR prefix. Current method IsValidOmrPrefix() is not compliant with that.   Separate from the valid OMR prefix question, a Domain Prefix is just never advertised in RIO because it's considered as on-link on the backbone link.
https://github.com/openthread/openthread/blob/main/src/core/border_router/routing_manager.cpp#L888

_Originally posted by @EskoDijk in https://github.com/openthread/openthread/issues/7900#issuecomment-1210523831_

abtink, 2022-08-11:
Thanks @EskoDijk .

- Submitted https://github.com/openthread/openthread/pull/8014 for this.

EskoDijk, 2022-08-11:
Ok, looks good - thanks.

closed 2022-08-19 by jwhui