openthread/openthread #8010: [routing-manager] classifying valid OMR prefixes

jwhui, 2022-08-10:
The method bool RoutingManager::IsValidOmrPrefix(const Ip6::Prefix &aOmrPrefix) looks overly restrictive. A valid OMR prefix could start with 0xfc as well. (Not necessarily 0xfd ) And a global IPv6 prefix does not necessarily start with the 3 bits 0b001 I think. It could also be configured in any other IPv6 namespace that may be made available in the future. I don't see value of a BR trying to 'play the role of a registry' and verify configured prefixes per today's IANA registrations. (Or is there a use case for that?) It seems not future proof.
https://github.com/openthread/openthread/blob/main/src/core/border_router/routing_manager.cpp#L894

_Originally posted by @EskoDijk in https://github.com/openthread/openthread/issues/7900#issuecomment-1210523831_

abtink, 2022-08-10:
Thanks @EskoDijk .

On the first point regarding ULA, I just submitted: https://github.com/openthread/openthread/pull/8012
- Looks like we already had a `Ip6::Prefix::IsUniqeLocal()` (which was checking for `fc00::/7`) but was not used.
- One key change here is that with this PR we also allow for any ULA prefix (not require the ULA to be 64 bit length) - which I think is correct behavior. Thoughts?

On the second point (GUA), I think we probably need more thoughts/discussions:
- Not clear to me how to define a GUA prefix?
   - We can go for https://www.rfc-editor.org/rfc/rfc4291#section-2.4 definition?
   - Or cover all reserved from https://www.iana.org/assignments/ipv6-address-space/ipv6-address-space.xhtml?
   -

EskoDijk, 2022-08-11:
In general for a ULA/GUA prefix, for it to be a valid OMR prefix, it is required that a Thread Device can perform SLAAC using the prefix per (6LoWPAN) SLAAC procedures.  RFC 4944 defines (https://www.rfc-editor.org/rfc/rfc4944.html#section-6) that the prefix length MUST be 64 bits for SLAAC. So both for GUA/ULA, 64 bits length is a requirement.

On the GUA, yes section 2.4 defines it and also https://www.rfc-editor.org/rfc/rfc4291.html#section-2.5.4 . So a GUA is roughly "everything else" and that's why we don't need to check much about it.

So the simplest way to check if a GUA or ULA is a valid OMR prefix is to check all of below:
* prefix length == 64
* prefix does not start with ff::/8
* prefix does not start with fe80::/10

EskoDijk, 2022-08-11:
PS so I think the ULA check as in #8012 isn't needed; and length check should be kept.  @jwhui Maybe you can verify if my RFC 4944 interpretation of a required /64 prefix to do SLAAC is correct.

jwhui, 2022-08-11:
> PS so I think the ULA check as in #8012 isn't needed; and length check should be kept. @jwhui Maybe you can verify if my RFC 4944 interpretation of a required /64 prefix to do SLAAC is correct.

[RFC 4944 Section 6](https://www.rfc-editor.org/rfc/rfc4944#section-6) does make the following statement:
```
   An IPv6 address prefix used for stateless autoconfiguration RFC4862
   of an IEEE 802.15.4 interface MUST have a length of 64 bits.
```

abtink, 2022-08-11:
Thanks guys!
- Submitted https://github.com/openthread/openthread/pull/8016 to change the check.
- Requiring prefix length to be 64 for sure keeps things simpler.

closed 2022-08-12 by jwhui