openthread/openthread #8026: When there are many packets sent to the thread, obtr is easy to lose packets and generate many errors such as: "failed: ChannelAccessFailure"

luyou-tuya, 2022-08-15:
**Describe the bug** A clear and concise description of what the bug is.
There are 17 matter thread sub-devices and an otbr device in the thread network. When the matter app sends a control command to each sub-device every 100ms, a report data will be reported when the status of the sub-device changes. It is easy to view the otbr log when the control is performed multiple times. Packets are lost and "failed: ChannelAccessFailure" and "error: NoBufs" will be generated. When I set the butterfly to a maximum of 8096, "error: NoBufs" disappears. This is because of the low performance of otbr or the low bandwidth of thread network. I think This amount of data is not large

**Mon Aug 15 17:36:23 2022 user.warn otbr-agent[2102]: 00:03:26.330 [W] Platform------: Handle transmit done failed: ChannelAccessFailure
Mon Aug 15 17:36:23 2022 user.notice otbr-agent[2102]: 00:03:26.330 [N] MeshForwarder-: Failed to send IPv6 UDP msg, len:82, chksum:851b, ecn:no, to:0x3000, sec:yes, error:ChannelAccessFailure, prio:low
Mon Aug 15 17:36:23 2022 user.notice otbr-agent[2102]: 00:03:26.330 [N] MeshForwarder-:     src:[fd7d:ce78:eb28:1:c102:7e93:4788:a98a]:5540
Mon Aug 15 17:36:23 2022 user.notice otbr-agent[2102]: 00:03:26.330 [N] MeshForwarder-:     dst:[fd7d:ce78:eb28:1:eab7:5560:ea7c:f04f]:5540
Mon Aug 15 17:36:25 2022 user.warn otbr-agent[2102]: 00:03:28.244 [W] Platform------: Handle transmit done failed: ChannelAccessFailure
Mon Aug 15 17:36:25 2022 user.notice otbr-agent[2102]: 00:03:28.244 [N] MeshForwarder-: Failed to send IPv6 UDP msg, len:82, chksum:dfd6, ecn:no, to:0x3000, sec:yes, error:ChannelAccessFailure, prio:low
Mon Aug 15 17:36:25 2022 user.notice otbr-agent[2102]: 00:03:28.244 [N] MeshForwarder-:     src:[fd7d:ce78:eb28:1:c102:7e93:4788:a98a]:5540
Mon Aug 15 17:36:25 2022 user.notice otbr-agent[2102]: 00:03:28.245 [N] MeshForwarder-:     dst:[fd7d:ce78:eb28:1:eab7:5560:ea7c:f04f]:5540
Mon Aug 15 17:36:26 2022 user.notice otbr-agent[2102]: 00:03:29.510 [N] MeshForwarder-: Failed to send IPv6 UDP msg, len:82, chksum:9ae2, ecn:no, to:0xbc00, sec:yes, error:NoAck, prio:low
Mon Aug 15 17:36:26 2022 user.notice otbr-agent[2102]: 00:03:29.510 [N] MeshForwarder-:     src:[fd7d:ce78:eb28:1:c102:7e93:4788:a98a]:5540
Mon Aug 15 17:36:26 2022 user.notice otbr-agent[2102]: 00:03:29.510 [N] MeshForwarder-:     dst:[fd7d:ce78:eb28:1:582a:644c:e864:afad]:5540
Mon Aug 15 17:36:28 2022 user.notice otbr-agent[2102]: 00:03:31.132 [N] MeshForwarder-: Failed to send IPv6 UDP msg, len:107, chksum:ce17, ecn:no, to:0x3000, sec:yes, error:NoAck, prio:low
Mon Aug 15 17:36:28 2022 user.notice otbr-agent[2102]: 00:03:31.133 [N] MeshForwarder-:     src:[fd7d:ce78:eb28:1:c102:7e93:4788:a98a]:5540
Mon Aug 15 17:36:28 2022 user.notice otbr-agent[2102]: 00:03:31.133 [N] MeshForwarder-:     dst:[fd7d:ce78:eb28:1:eab7:5560:ea7c:f04f]:5540
Mon Aug 15 17:36:28 2022 user.warn otbr-agent[2102]: 00:03:31.652 [W] Platform------: Handle transmit done failed: ChannelAccessFailure
Mon Aug 15 17:36:28 2022 user.notice otbr-agent[2102]: 00:03:31.652 [N] MeshForwarder-: Failed to send IPv6 UDP msg, len:82, chksum:ba97, ecn:no, to:0xac00, sec:yes, error:ChannelAccessFailure, prio:low
Mon Aug 15 17:36:28 2022 user.notice otbr-agent[2102]: 00:03:31.652 [N] MeshForwarder-:     src:[fd7d:ce78:eb28:1:c102:7e93:4788:a98a]:5540
Mon Aug 15 17:36:28 2022 user.notice otbr-agent[2102]: 00:03:31.653 [N] MeshForwarder-:     dst:[fd7d:ce78:eb28:1:3d78:6a90:f97b:790e]:5540
Mon Aug 15 17:36:34 2022 user.warn otbr-agent[2102]: 00:03:37.772 [W] Platform------: Failed to write CLI output: Broken pipe
Mon Aug 15 17:36:41 2022 user.notice otbr-agent[2102]: 00:03:44.260 [N] MeshForwarder-: Failed to send IPv6 UDP msg, len:107, chksum:184d, ecn:no, to:0xbc00, sec:yes, error:NoAck, prio:low
Mon Aug 15 17:36:41 2022 user.notice otbr-agent[2102]: 00:03:44.260 [N] MeshForwarder-:     src:[fd7d:ce78:eb28:1:c102:7e93:4788:a98a]:5540
Mon Aug 15 17:36:41 2022 user.notice otbr-agent[2102]: 00:03:44.260 [N] MeshForwarder-:     dst:[fd7d:ce78:eb28:1:582a:644c:e864:afad]:5540
Mon Aug 15 17:36:42 2022 user.warn otbr-agent[2102]: 00:03:45.548 [W] Platform------: Daemon read: Connection reset by peer
Mon Aug 15 17:36:42 2022 user.notice otbr-agent[2102]: 00:03:45.777 [N] MeshForwarder-: Failed to send IPv6 UDP msg, len:82, chksum:7f35, ecn:no, to:0xb800, sec:yes, error:NoAck, prio:low
Mon Aug 15 17:36:42 2022 user.notice otbr-agent[2102]: 00:03:45.777 [N] MeshForwarder-:     src:[fd7d:ce78:eb28:1:c102:7e93:4788:a98a]:5540
Mon Aug 15 17:36:42 2022 user.notice otbr-agent[2102]: 00:03:45.777 [N] MeshForwarder-:     dst:[fd7d:ce78:eb28:1:9e7b:43a3:ff02:826a]:5540
Mon Aug 15 17:36:46 2022 user.notice otbr-agent[2102]: 00:03:49.133 [N] MeshForwarder-: Failed to send IPv6 UDP msg, len:90, chksum:766a, ecn:no, to:0xac00, sec:yes, error:NoAck, prio:low
Mon Aug 15 17:36:46 2022 user.notice otbr-agent[2102]: 00:03:49.133 [N] MeshForwarder-:     src:[fd7d:ce78:eb28:1:c102:7e93:4788:a98a]:5540
Mon Aug 15 17:36:46 2022 user.notice otbr-agent[2102]: 00:03:49.133 [N] MeshForwarder-:     dst:[fd7d:ce78:eb28:1:206d:ad5b:8517:106a]:5540
Mon Aug 15 17:36:46 2022 user.warn otbr-agent[2102]: 00:03:49.354 [W] Platform------: Failed to write CLI output: Broken pipe
Mon Aug 15 17:36:48 2022 user.warn otbr-agent[2102]: 00:03:51.608 [W] Platform------: Failed to write CLI output: Broken pipe
Mon Aug 15 17:36:49 2022 user.warn otbr-agent[2102]: 00:03:52.552 [W] Platform------: Handle transmit done failed: ChannelAccessFailure
Mon Aug 15 17:36:49 2022 user.notice otbr-agent[2102]: 00:03:52.553 [N] MeshForwarder-: Failed to send IPv6 UDP msg, len:82, chksum:f069, ecn:no, to:0xf400, sec:yes, error:ChannelAccessFailure, prio:low
Mon Aug 15 17:36:49 2022 user.notice otbr-agent[2102]: 00:03:52.554 [N] MeshForwarder-:     src:[fd7d:ce78:eb28:1:c102:7e93:4788:a98a]:5540
Mon Aug 15 17:36:49 2022 user.notice otbr-agent[2102]: 00:03:52.554 [N] MeshForwarder-:     dst:[fd7d:ce78:eb28:1:3df0:4a39:6b61:5fa8]:5540
Mon Aug 15 17:36:50 2022 user.warn otbr-agent[2102]: 00:03:53.783 [W] Platform------: Daemon read: Connection reset by peer
Mon Aug 15 17:36:52 2022 user.notice otbr-agent[2102]: 00:03:55.653 [N] MeshForwarder-: Failed to send IPv6 UDP msg, len:90, chksum:8059, ecn:no, to:0x3400, sec:yes, error:NoAck, prio:low
Mon Aug 15 17:36:52 2022 user.notice otbr-agent[2102]: 00:03:55.653 [N] MeshForwarder-:     src:[fd7d:ce78:eb28:1:c102:7e93:4788:a98a]:5540
Mon Aug 15 17:36:52 2022 user.notice otbr-agent[2102]: 00:03:55.653 [N] MeshForwarder-:     dst:[fd7d:ce78:eb28:1:2432:5c4a:20:c3c3]:5540
Mon Aug 15 17:37:01 2022 user.warn otbr-agent[2102]: 00:04:04.953 [W] Platform------: Failed to write CLI output: Broken pipe
Mon Aug 15 17:37:09 2022 user.warn otbr-agent[2102]: 00:04:12.161 [W] Platform------: Handle transmit done failed: ChannelAccessFailure
Mon Aug 15 17:37:09 2022 user.notice otbr-agent[2102]: 00:04:12.162 [N] MeshForwarder-: Failed to send IPv6 UDP msg, len:82, chksum:f3bf, ecn:no, to:0x3000, sec:yes, error:ChannelAccessFailure, prio:low
Mon Aug 15 17:37:09 2022 user.notice otbr-agent[2102]: 00:04:12.163 [N] MeshForwarder-:     src:[fd7d:ce78:eb28:1:c102:7e93:4788:a98a]:5540
Mon Aug 15 17:37:09 2022 user.notice otbr-agent[2102]: 00:04:12.164 [N] MeshForwarder-:     dst:[fd7d:ce78:eb28:1:eab7:5560:ea7c:f04f]:5540
Mon Aug 15 17:37:10 2022 user.warn otbr-agent[2102]: 00:04:13.094 [W] Platform------: Daemon read: Connection reset by peer
Mon Aug 15 17:37:13 2022 user.warn otbr-agent[2102]: 00:04:16.435 [W] Platform------: Daemon read: Connection reset by peer
Mon Aug 15 17:37:18 2022 user.warn otbr-agent[2102]: 00:04:21.340 [W] Platform------: Failed to write CLI output: Broken pipe
Mon Aug 15 17:37:20 2022 user.warn otbr-agent[2102]: 00:04:23.181 [W] Platform------: Failed to write CLI output: Broken pipe
Mon Aug 15 17:37:33 2022 user.warn otbr-agent[2102]: 00:04:36.369 [W] Platform------: Daemon read: Connection reset by peer
Mon Aug 15 17:37:36 2022 user.notice otbr-agent[2102]: 00:04:39.800 [N] MeshForwarder-: Dropping (reassembly queue) IPv6 UDP msg, len:816, chksum:809f, ecn:no, sec:yes, error:ReassemblyTimeout, prio:normal, rss:-76.0
Mon Aug 15 17:37:36 2022 user.notice otbr-agent[2102]: 00:04:39.801 [N] MeshForwarder-:     src:[fd7d:ce78:eb28:1:582a:644c:e864:afad]:5540
Mon Aug 15 17:37:36 2022 user.notice otbr-agent[2102]: 00:04:39.801 [N] MeshForwarder-:     dst:[fd7d:ce78:eb28:1:c102:7e93:4788:a98a]:5540
[... cut ...]
Mon Aug 15 18:17:33 2022 user.notice otbr-agent[2102]: 00:44:36.641 [N] MeshForwarder-:     dst:[fd7d:ce78:eb28:1:582a:644c:e864:afad]:5540
Mon Aug 15 18:17:36 2022 user.notice otbr-agent[2102]: 00:44:39.188 [N] MeshForwarder-: Failed to send IPv6 UDP msg, len:90, chksum:6afc, ecn:no, to:0x0800, sec:yes, error:NoAck, prio:low
Mon Aug 15 18:17:36 2022 user.notice otbr-agent[2102]: 00:44:39.188 [N] MeshForwarder-:     src:[fd7d:ce78:eb28:1:c102:7e93:4788:a98a]:5540
Mon Aug 15 18:17:36 2022 user.notice otbr-agent[2102]: 00:44:39.189 [N] MeshForwarder-:     dst:[fd7d:ce78:eb28:1:8d70:4f5a:9e1e:c5ca]:5540

luyou-tuya, 2022-08-16:
> 
no! It is to send a control command to a sub-device every 100ms, 10 commands per second, until 17 devices are sent, but each control command of matter will have a status response and ack, and the app subscribes to the properties of the sub-device. , the control command changes the state of the attribute, the sub-device will report two pieces of data at the same time, and the reported piece of data also requires the app to reply a status response and ack, such a control command will generate a maximum of 12 data packets, sometimes ack and response data Reply together so that at least 9 packets are generated, Then send the 17 commands again on a manual click, and repeat.
**IEEE 802.15.4 hardware platform**
otbr is Silicon Labs efr32mg21 
sub-device Silicon Labs efr32mg24A020F1536
**Git commit id**
otbr: 3a98779dc
openthread: 3a98779dc
**Network topology**
Sorry！ I don't know how to provide

luyou-tuya, 2022-08-16:
> @luyou-tuya , thanks for raising this issue.
> 
> If I understand properly, your test involves performing a Matter interaction to each of 17 Matter devices every 100 ms. In other words, that means performing a Matter interaction 170 times per second. And because a Matter interaction typically involves a request + response, that is really 340 packets per second. Is that correct? If so, Thread is not really able to handle a traffic load of 340 packets/second. Note that the Thread PHY supports 250 kilobits/second. With MAC and header overhead, that typically reduces usable bandwidth to about half. Multihop routing alone will also reduce to one-third, so then you're down to something closer of 42 kbps in the best case.
> 
> Can you also help provide more information about your setup?
> 
> * Git commit id
> * IEEE 802.15.4 hardware platform
> * Network topology

no! It is to send a control command to a sub-device every 100ms, 10 commands per second, until 17 devices are sent, but each control command of matter will have a status response and ack, and the app subscribes to the properties of the sub-device. , the control command changes the state of the attribute, the sub-device will report two pieces of data at the same time, and the reported piece of data also requires the app to reply a status response and ack, such a control command will generate a maximum of 12 data packets, sometimes ack and response data Reply together so that at least 9 packets are generated, Then send the 17 commands again on a manual click, and repeat.
IEEE 802.15.4 hardware platform
otbr is Silicon Labs efr32mg21
sub-device Silicon Labs efr32mg24A020F1536
Git commit id
otbr: 3a98779dc
openthread: 3a98779dc
Network topology
Sorry！ I don't know how to provide

jwhui, 2022-08-16:
> no! It is to send a control command to a sub-device every 100ms, 10 commands per second, until 17 devices are sent, but each control command of matter will have a status response and ack, and the [...]

So 10 commands per second with 9-12 packets per command, that's 90-120 packets per second? Is that correct? That's still relatively high traffic load for Thread.

Can you provide a packet trace for performing a command with a single node, just so we can have better visibility into the exact traffic you are trying to describe?

/cc/ @romacdon @suveshpratapa

luyou-tuya, 2022-08-16:
> So 10 commands per second with 9-12 packets per command, that's 90-120 packets per second? Is that correct? That's still relatively high traffic load for Thread.
>
> Can you provide a packet trace for [...]

ok ! I used wireshark to capture the packets and the logs of otbr

luyou-tuya, 2022-08-16:
> So 10 commands per second with 9-12 packets per command, that's 90-120 packets per second? Is that correct? That's still relatively high traffic load for Thread.
>
> Can you provide a packet trace for [...]

luyou-tuya, 2022-08-16:
> So 10 commands per second with 9-12 packets per command, that's 90-120 packets per second? Is that correct? That's still relatively high traffic load for Thread.
>
> Can you provide a packet trace for [...]

The file name 123467.pcap is to capture the data traffic packet flow of the network card name wpan0 for one minute.
The file name one_device_one_command_packet.pcap is the data packet generated by a sub-device and a command without retransmission
[data_pcap.zip](https://github.com/openthread/openthread/files/9349867/data_pcap.zip)

closed 2022-08-16 by luyou-tuya