openthread/openthread #8037: Why return two SRP services with the same information?

lancersky, 2022-08-17:
**Describe the bug** A clear and concise description of what the bug is.
```
# ot-ctl srp server service
aaa-bbb._knx._udp.default.service.arpa.
    deleted: false
    subtypes: (null)
    port: 5683
    priority: 0
    weight: 0
    ttl: 3600
    TXT: []
    host: bbb.default.service.arpa.
    addresses: [fd11:22:0:0:996f:d29f:1474:f08a]
aaa-bbb._knx._udp.default.service.arpa.
    deleted: false
    subtypes: (null)
    port: 5683
    priority: 0
    weight: 0
    ttl: 3600
    TXT: []
    host: bbb.default.service.arpa.
    addresses: [fd11:22:0:0:996f:d29f:1474:f08a]
Done
```
**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id 
https://github.com/openthread/ot-br-posix/commit/3a98779dc9f0875af9c7b137e764c9663bfdc9c3
[mdnsresponder - 1310.80.1](https://opensource.apple.com/tarballs/mDNSResponder/mDNSResponder-1310.80.1.tar.gz)
2. IEEE 802.15.4 hardware platform
4. Build steps
5. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

wgtdkp, 2022-08-23:
@lancersky Can you provide the SRP client software version and the steps to reproduce the error? Also, better if you could capture the Thread network traffic.

------

Looks similar to https://github.com/openthread/ot-br-posix/issues/1425

lancersky, 2022-08-23:
Yeah, It's the same issue with https://github.com/openthread/ot-br-posix/issues/1425

abtink, 2022-08-24:
@lancersky I added a comment on https://github.com/openthread/ot-br-posix/issues/1425#issuecomment-1225305561 with some findings.

abtink, 2022-08-24:
I think we found the root cause (see discussion in #1425). Quick summary:

- SRP client seems to  use incorrect sub-type naming model (causing two similarly named services to be registered).
   - SRP client used is not the OpenThread implementation.
- The names happen to be chosen such that the name check in OT SRP server pass (both service names are suffixes of the service instance name).
- We can later add additional checks in OT SRP server to reject such mallformed names.
- But the immediate fix should be on the SRP client side (which is not the OpenThread implementation).

closed 2022-08-25 by jwhui