openthread/openthread #8046: [github-actions] test_multi_border_routers.py fails intermittently

jwhui, 2022-08-19:
```
======================================================================
FAIL: test (__main__.Nat64MultiBorderRouter)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "./tests/scripts/thread-cert/border_router/nat64/test_multi_border_routers.py", line 137, in test
    self.assertEqual(nat64_prefix, br1_local_nat64_prefix)
AssertionError: 'fd1d:206:bedd:2:0:0::/96' != 'fdb4:e654:e221:2:0:0::/96'
- fd1d:206:bedd:2:0:0::/96
+ fdb4:e654:e221:2:0:0::/96

----------------------------------------------------------------------
```

https://github.com/openthread/openthread/runs/7912421414?check_suite_focus=true

xuyirio, 2022-08-22:
Fixing in [PR#8057](https://github.com/openthread/openthread/pull/8057).

jwhui, 2022-08-22:
Resolved by #8057

closed 2022-08-22 by jwhui