openthread/openthread #8048: MLE Link Request cannot be transmitted immediately when REED upgrades to Active Router

HuangCC-IFX, 2022-08-19:
**Describe the bug** To synchronize the link between new Active Router (upgraded from an existing REED) and existent Routers/Leader, the new Active Router shall issue a multicast MLE Link Request frame to the network before accepting the connecting end device. However, in @jwhui 's commit, https://github.com/openthread/openthread/pull/7947/files, the MLE Link Request will not be transmitted immediately after receiving the Address Solicit Response. 

**To Reproduce** : Thread 1.1 Test Case 5.2.4.

1. Git commit id: f75d03f6cbd77b3d9ca0a630934114f12560b3fc, 0b1e922a1d446e58daaafdbf4840ddfbfd80dd73
2. IEEE 802.15.4 hardware platform: Infineon CYW30739
3. Build steps
4. Network topology: Refer to Thread 1.1 Test Case 5.2.4

**Expected behavior** The joining end device shall not be accepted (by issuing the MLE Child Id Response) before link is synchronized.
[image]

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

jwhui, 2022-08-24:
@HuangCC-IFX , thanks for raising this issue.

The current MLE Link Request behavior was introduced in #7745 based on discussion in #7359. Can you point to anything in the Thread Specification that requires sending an MLE Link Request immediately after receiving an Address Solicit Response?

I also think it is valid for a newly upgraded router to accept children even before it has established links with neighboring routers. Can you point to anything in the Thread Specification that states otherwise?

HuangCC-IFX, 2022-08-26:
@jwhui 
Actually, there is no explicit definition for the order of Link Request and Child ID Responder frames.
But there is a requirement for REED becomes an Active Router in Section 4.7.1 in Thread v1.3.0 spec..
============================================
Link synchronization MUST be performed in the following circumstances:
- After the attachment process described in Section 4.5.1, Attaching to a Parent
- **When a REED becomes an Active Router.**
- When a Router receives a message from a neighboring Device for which it does not have a stored frame counter.
- After a hardware reset or other event that causes a loss of local data and/or allows an indeterminate amount of time to pass.
============================================

In my consideration, we shall first determine if the REED can have its child device.

1. If not, the REED shall first upgrade to Active Router and then can accept the Child ID Request by issuing the corresponding Child ID Response. To meet this requirement, at least one Link synchronization process (either 2-message exchange or 3-messages ) shall be accomplished. Otherwise, the new Active Router (upgraded from the existent REED) can NOT ensure its routing in the network.
2. If yes, the REED is not required to be upgraded to Active Router.

jwhui, 2022-08-26:
@HuangCC-IFX , immediately sending an MLE Link Request after receiving an Address Solicit Response does not work in the general case. Prior to #7745 , we did send the multicast Link Request immediately. However, the problem with doing that is neighboring routers may not have received the required update from the Leader that a new Router ID has been allocated, as a result the multicast Link Request is ignored by all neighboring routers. Our solution to this is by waiting to receive an MLE Advertisement message from at least one neighboring router indicating that it has received the update from the Leader before sending a multicast MLE Link Request. See the discussion in #7359 for further details.

HuangCC-IFX, 2022-08-26:
@jwhui 
I agree with your consideration for #7359. But in #7359, there is only one link between the REED (node A) and Router (node B).
In my previous reply, I think the REED can be upgraded to Active Router ONLY at least one Link Synchronization process is accomplished. Hence, I suggest to add a timeout timer and a state machine to handle the Child Join Request. That is, if there is no available link established for this new Active Router before timeout, the new Active Router shall reject the Child Join Request and downgrade to REED again instead of issuing the Child ID Response immediately after receiving the Acknowledgement of Address Solicit Request.

jwhui, 2022-08-26:
@HuangCC-IFX , simply establishing one link with a neighboring router does not by itself ensure routing to the rest of the Thread network. The routing topology still needs to converge, which can require a number of MLE Advertisement exchanges. And even with a routing topology established, that does not guarantee reachability to a Leader. Already active Thread routers do not stop sending Child ID Response messages when they are temporarily disconnected from neighboring routers and/or the Leader.

Can you describe an issue you are seeing with real-world testing and/or deployments that would be addressed by delaying Child ID Response messages? Ideally, if you can share any real-world testing data to show the benefits of your proposed change. I think it would help if we better understand the use case. As I mentioned before, there are a number of reasons where an active router may only provide various levels of connectivity at any given point in time.

HuangCC-IFX, 2022-08-26:
@jwhui 

1. How to attach/upload my test result for TC 5.2.4 in the github?
2. I agree that the routing path varies with the real air status. I only emphasize to ensure at least one link is established for the new Active Router to exchange data in the existent network. Actually, no one can guarantee the reachability to the other node in every moment. Network rescure replies on every node in the mesh network. Although in my test, the network works well even the link synchronization has been accomplished later, the Test Harness judges the test as fail since the Child ID Response is sent in prior to the Link Request frame. After investigating carefully, I think that somewhat make sense.

jwhui, 2022-08-26:
@HuangCC-IFX , if this is related to the Thread Test Harness, I would suggest filing a Thread Group JIRA (DEV or TESTPLAN) against the test harness so that we can discuss within the Thread Group.

HuangCC-IFX, 2022-09-07:
@jwhui 
The JIRA was created. https://threadgroup.atlassian.net/browse/TESTPLAN-527

jwhui, 2022-10-20:
[TESTPLAN-527](https://threadgroup.atlassian.net/browse/TESTPLAN-527) has been resolved.

closed 2022-10-20 by jwhui