openthread/openthread #8056: How to solve 'SrpServer-----: Name conflict'

lancersky, 2022-08-22:
**Describe the bug** A clear and concise description of what the bug is.
If SRP Client re-register, it will give a name conflict error in SRP Server.
So, How do I properly update the SRP Client information when my end device IP changes after reboot?
```
Mon Aug 22 09:59:21 2022 user.warn otbr-agent[21919]: 0d.18:00:11.885 [W] SrpServer-----: Name conflict: host name f4ce3614c96b8b4d.default.service.arpa. has already been allocated
Mon Aug 22 09:59:21 2022 user.warn otbr-agent[21919]: 0d.18:00:11.886 [W] SrpServer-----: Failed to process DNS Update section: Duplicated
Mon Aug 22 09:59:21 2022 user.warn otbr-agent[21919]: 0d.18:00:11.886 [W] SrpServer-----: Send fail response: 6
Mon Aug 22 09:59:23 2022 user.warn otbr-agent[21919]: 0d.18:00:13.707 [W] SrpServer-----: Name conflict: host name f4ce3614c96b8b4d.default.service.arpa. has already been allocated
Mon Aug 22 09:59:23 2022 user.warn otbr-agent[21919]: 0d.18:00:13.708 [W] SrpServer-----: Failed to process DNS Update section: Duplicated
Mon Aug 22 09:59:23 2022 user.warn otbr-agent[21919]: 0d.18:00:13.708 [W] SrpServer-----: Send fail response: 6
Mon Aug 22 09:59:26 2022 user.warn otbr-agent[21919]: 0d.18:00:16.796 [W] SrpServer-----: Name conflict: host name f4ce3614c96b8b4d.default.service.arpa. has already been allocated
Mon Aug 22 09:59:26 2022 user.warn otbr-agent[21919]: 0d.18:00:16.797 [W] SrpServer-----: Failed to process DNS Update section: Duplicated
Mon Aug 22 09:59:26 2022 user.warn otbr-agent[21919]: 0d.18:00:16.798 [W] SrpServer-----: Send fail response: 6
Mon Aug 22 09:59:32 2022 user.warn otbr-agent[21919]: 0d.18:00:22.033 [W] SrpServer-----: Name conflict: host name f4ce3614c96b8b4d.default.service.arpa. has already been allocated
Mon Aug 22 09:59:32 2022 user.warn otbr-agent[21919]: 0d.18:00:22.034 [W] SrpServer-----: Failed to process DNS Update section: Duplicated
Mon Aug 22 09:59:32 2022 user.warn otbr-agent[21919]: 0d.18:00:22.035 [W] SrpServer-----: Send fail response: 6
Mon Aug 22 09:59:40 2022 user.warn otbr-agent[21919]: 0d.18:00:30.915 [W] SrpServer-----: Name conflict: host name f4ce3614c96b8b4d.default.service.arpa. has already been allocated
Mon Aug 22 09:59:40 2022 user.warn otbr-agent[21919]: 0d.18:00:30.916 [W] SrpServer-----: Failed to process DNS Update section: Duplicated
Mon Aug 22 09:59:40 2022 user.warn otbr-agent[21919]: 0d.18:00:30.916 [W] SrpServer-----: Send fail response: 6

root@OpenWrt:~# ot-ctl srp server service
my_instance_name._ipps._udp.default.service.arpa.
    deleted: true
Done
```

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id https://github.com/openthread/ot-br-posix/commit/3a98779dc9f0875af9c7b137e764c9663bfdc9c3
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

jwhui, 2022-08-22:
SRP implements First-Come, First Served naming. To prevent name conflicts, SRP clients maintain a key to sign SRP updates. As long as the SRP Client maintains the same key, it should be able to send SRP Updates (i.e. re-register) with the same name(s) that it had successfully registered in the past:
https://github.com/openthread/openthread/blob/d48799dddfde76b7b228f07144ba1d8f4e08e2e6/src/core/net/srp_server.cpp#L339

The SRP Client should maintain the SRP Key in non-volatile storage:
https://github.com/openthread/openthread/blob/d48799dddfde76b7b228f07144ba1d8f4e08e2e6/src/core/common/settings.hpp#L118

If the key is lost for whatever reason (including a factory reset) the SRP Client will not be able to re-register the same name(s).

wgtdkp, 2022-08-23:
@lancersky Can you clarify if below is true for your SRP client device: the device has been factory reset but the same value is used to construct the SRP host name `f4ce3614c96b8b4d.default.service.arpa`?

Factoryreset changes the SRP client key and name conflicts happen when the same name is registered with different keys. Note that
```
root@OpenWrt:~# ot-ctl srp server service
my_instance_name._ipps._udp.default.service.arpa.
    deleted: true
Done
```
indicates the DNS Resource Records for the service are deleted but the name is still allocated for the original client (you should check the host instead by the way `ot-ctl srp server host`)

jwhui, 2022-09-23:
Closing stale issue.

closed 2022-09-23 by jwhui