openthread/openthread #8066: Compile error (warning) on Linux

dtodor, 2022-08-23:
The compiler (GCC 11.2.0) is apparently not smart enough to figure out that `hints` will be initialised in case of an error.

https://github.com/openthread/openthread/blame/main/src/posix/platform/infra_if.cpp#L647

```
third_party/openthread/repo/src/posix/platform/infra_if.cpp
third_party/openthread/repo/src/posix/platform/infra_if.cpp: In member function ‘otError ot::Posix::InfraNetif::DiscoverNat64Prefix(uint32_t)’:
third_party/openthread/repo/src/posix/platform/infra_if.cpp:647:21: error: ‘hints’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  647 |         freeaddrinfo(hints);
      |         ~~~~~~~~~~~~^~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
```

jwhui, 2022-08-23:
I just submitted #8070 to fix the issue.

dtodor, 2022-08-24:
OK, but now you are freeing an `nullptr`. Is this safe?

PR #8070 (2022-08-25): [posix] initialize `hints` in `InfraNetif::DiscoverNat64Prefix`
Fixes #8066
closed 2022-08-25 by jwhui