openthread/openthread #8074: OTBR can't

wallaceL, 2022-08-24:
I have two otbrs in the same lan. One otbr named A send RA(PIO & RIO),  so the another otbr named B and  devices in the same lan get IPV6 address. If I power off otbr A,  incoming device joined this lan, can't get IPV6 address until otbr B send RA with PIO info, but it need long time.  to reduce stale time(kRtrAdvStaleTime) ? 

1. commit id: 8f92d2dc8
3. IEEE 802.15.4 hardware platform
4. Build steps
5. Network topology

jwhui, 2022-08-24:
@wallaceL , thanks for raising this issue. This is an issue that I have noted as needing to be addressed.

OpenThread currently tracks usable prefixes advertised by other BRs on the infrastructure link. I think we should add an enhancement where the BR is actively verifying that BRs advertising usable prefixes are still available. We can do this by periodically sending IPv6 Neighbor Solicitation message to verify reachability of a neighboring BR. If a neighboring BR does not respond to a Neighbor Solicitation after a few retries, then we can declare the prefixes it was advertising as unusable.

Thoughts?

abtink, 2022-08-25:
I think it make sense to add a mechanism to check reachability with neighboring BR/router (using unicast NS and/or RS msgs). We can consider adding this later.

In addition to this, just submitted this PR
- https://github.com/openthread/openthread/pull/8079
- This helps adds redundancy by allowing multiple BRs advertise the same local on-link prefix in their emitted RAs (if it is determined as the winner/favored on-link prefix).
- On a Thread BR the local on-link prefix is derived from Extended PAN ID and so it will be the same (on all BRs connected to same Thread mesh).

jwhui, 2022-08-25:
Thanks, @abtink . Allowing multiple BRs from the same Thread network will certainly help with redundancy. At the same time, I think it will be useful to periodically validate reachability of routers that are currently advertising usable prefixes. Such routers may be from other Thread networks or may not be related to Thread at all.

abtink, 2022-08-27:
Submitted https://github.com/openthread/openthread/pull/8090 related to this.

closed 2022-10-25 by jwhui