openthread/openthread #8077: Cannot communicate with End device using the ML-EID when enable OPENTHREAD_SLAAC

lancersky, 2022-08-24:
**Describe the bug** A clear and concise description of what the bug is.
I created a network on OTBR.
```
###OTBR
root@OpenWrt:~# ot-ctl dataset active
Active Timestamp: 0
Channel: 26
Channel Mask: 0x07fff800
Ext PAN ID: cc9db289d8483c12
Mesh Local Prefix: fd60:8767:772a:c3a4::/64
Network Key: 8f3a0d3d201044f830104cd18ecb97bd
Network Name: OpenThread-9879
PAN ID: 0x9879
PSKc: 5d2e31e4c385a75d7607b8fe99a95398
Security Policy: 672 onrc
Done
```
When disable OPENTHREAD_SLAAC on End device, the OTBR can communicate with End device using the ML-EID.
[disable CONFIG_OPENTHREAD_SLAAC.zip](https://github.com/openthread/openthread/files/9415587/disable.CONFIG_OPENTHREAD_SLAAC.zip)
```
###End Device
uart:~$ ot ipaddr
fd60:8767:772a:c3a4:0:ff:fe00:7c0e
fd60:8767:772a:c3a4:cd82:4e9b:281a:3869
fe80:0:0:0:c877:542e:850f:36d2
Done
uart:~$ ot netdata show
Prefixes:
fd11:22:0:0::/64 paros med 7c00
Routes:
Services:
44970 5d fd608767772ac3a4c5b90a4fe2b64a3fd120 s 7c00
44970 01 4304b000000e10 s 7c00
Done
```
Use the following command to get the correct response.
```
###OTBR
root@OpenWrt:~# coap-client -N -e 0 -m put coap://[fd60:8767:772a:c3a4:cd82:4e9b:281a:3869]/light
hello

###End Device
uart:~$ 
[00:02:41.902,893] <inf> ot_coap_utils: Received light request: 0
[00:02:41.903,076] <inf> ot_coap_utils: Sent provisioning response:
                                        68 65 6c 6c 6f                                   |hello    
```

But when enable OPENTHREAD_SLAAC on End device,  the OTBR cannot communicate with End device using the ML-EID.
[enable CONFIG_OPENTHREAD_SLAAC.zip](https://github.com/openthread/openthread/files/9415596/enable.CONFIG_OPENTHREAD_SLAAC.zip)
```
###End Device
uart:~$ ot ipaddr
fd60:8767:772a:c3a4:0:ff:fe00:e000
fd11:22:0:0:d8a9:fbdc:39b6:f412
fd60:8767:772a:c3a4:d8ce:f172:c2ca:70a7
fe80:0:0:0:c1:2248:a974:effa
Done
uart:~$ ot netdata show
Prefixes:
fd11:22:0:0::/64 paros med 7c00
Routes:
Services:
44970 5d fd608767772ac3a4c5b90a4fe2b64a3fd120 s 7c00
44970 01 4304b000000e10 s 7c00
Done
```
Using the following command does not get the response. But the End device receives the message correctly.
```
###OTBR
root@OpenWrt:~# ot-ctl ping fd60:8767:772a:c3a4:d8ce:f172:c2ca:70a7
16 bytes from fd60:8767:772a:c3a4:d8ce:f172:c2ca:70a7: icmp_seq=12 hlim=64 time=14ms
1 packets transmitted, 1 packets received. Packet loss = 0.0%. Round-trip min/avg/max = 14/14.0/14 ms.
Done
root@OpenWrt:~# ping -6 -c1 fd60:8767:772a:c3a4:d8ce:f172:c2ca:70a7
PING fd60:8767:772a:c3a4:d8ce:f172:c2ca:70a7 (fd60:8767:772a:c3a4:d8ce:f172:c2ca:70a7): 56 data bytes
64 bytes from fd60:8767:772a:c3a4:d8ce:f172:c2ca:70a7: seq=0 ttl=64 time=19.503 ms

--- fd60:8767:772a:c3a4:d8ce:f172:c2ca:70a7 ping statistics ---
1 packets transmitted, 1 packets received, 0% packet loss
round-trip min/avg/max = 19.503/19.503/19.503 ms
root@OpenWrt:~# coap-client -N -e 0 -m put coap://[fd60:8767:772a:c3a4:d8ce:f172:c2ca:70a7]/light
^Croot@OpenWrt:~# 

###End Device
[00:38:33.900,726] <inf> ot_coap_utils: Received light request: 0
[00:38:33.900,909] <inf> ot_coap_utils: Sent provisioning response:
                                        68 65 6c 6c 6f                                   |hello   
```
If use On-mesh IP address, it work well. 
[enable CONFIG_OPENTHREAD_SLAAC and use on-mesh ipaddr.zip](https://github.com/openthread/openthread/files/9415603/enable.CONFIG_OPENTHREAD_SLAAC.and.use.on-mesh.ipaddr.zip)
```
###OTBR
root@OpenWrt:~# coap-client -N -e 0 -m put coap://[fd11:22:0:0:d8a9:fbdc:39b6:f412]/light
hello
root@OpenWrt:~# 

###End Device
[00:42:02.894,256] <inf> ot_coap_utils: Received light request: 0
[00:42:02.894,439] <inf> ot_coap_utils: Sent provisioning response:
                                        68 65 6c 6c 6f                                   |hello  
```

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id https://github.com/openthread/ot-br-posix/commit/3a98779dc9f0875af9c7b137e764c9663bfdc9c3
https://github.com/nrfconnect/sdk-nrf/commit/ffd959dbc549e142f1785eec136559b8d7a87947
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

LuDuda, 2022-08-25:
Hello @lancersky

In general, I don't recommend using the ML-EID address for such communication. Using On-Mesh prefix (`fd11:22..` in your case) would allow also other nodes connected to the same network as your OpenWRT router to reach the Thread device.

I see that you have modified the `coap-server` example, though I think I was able to reproduce your finding.

Basically, the `coap-client` application from `libcoap` expects to receive the response from the same IPv6 address to which the request was made (internally it creates a hash out of it). Otherwise, the response is dropped.

I believe in your code you have cleared the `mSockAddr` from `otMessageInfo` before sending the response.
`memset(&msg_info.mSockAddr, 0, sizeof(msg_info.mSockAddr));`

Now, when SLAAC is disabled only the ML-EID address is available on the node. The `coap-client` sends the CoAP message to the ML-EID and receives a reply also from the ML-EID - OK.

When SLAAC is enabled we have both ML-EID and On-Mesh addresses on the device. The `coap-client` may send the CoAP message to ML-EID or On-Mesh addresses, but as the `mSockAddr` is cleared, it replies with the address selected by the IPv6 stack. Since `coap-client` initiates a request from source address constructed from On-Mesh prefix, the device always chooses its own On-Mesh address as a best match. All in all, it is expected that sending request to the ML-EID address from `coap-client` will not show any response. To fix it, just remove the mentioned `memset`.

lancersky, 2022-08-26:
Thanks for your reply. It solves my problem.

closed 2022-08-26 by lancersky