openthread/openthread #8082: If there are many thread sub-devices, the srp service of some sub-devices cannot be registered when otbr is restarted

luyou-tuya, 2022-08-25:
There are 37 sub-devices and one otbr in the thread network. When otbr is restarted, the otbr information of some thread's sub-devices cannot be registered. After otbr restarts, there is no other control data traffic in the outside world. At this time, only all sub-devices will send to otbr. srp service, which will cause the thread network congestion, check the log information of otbr, and find that the data packets discarded by the udp data reorganization timeout will be lost, and the later received fragmented transmission data will also be lost.
Sub-devices that fail to register with the srp service cannot be registered after a long period of time. Sometimes, all sub-device srp information can be seen after a period of time. I see that the code of srpclient will send SRP between the time interval of kMinRetryWaitInterval~kMaxRetryWaitInterval update message.
What is the reason at this time, have you tested this situation?

EEE 802.15.4 hardware platform
otbr is Silicon Labs efr32mg21
sub-device Silicon Labs efr32mg24A020F1536
Git commit id
otbr: 3a98779dc
openthread: https://github.com/openthread/openthread/commit/8f92d2dc81243f62d6cd609956cdb0120273272d

otbr log:
Fri Jan  2 09:59:43 1970 user.warn otbr-agent[2080]: 01:32:53.262 [W] Platform------: [netif] Successfully deleted an external route fd58:bb64:ce7:1::/64 in kernel
Fri Jan  2 09:59:44 1970 user.warn otbr-agent[2080]: [WARN]-MDNS----: Failed to send goodbye message for host 0E1869581A79DFC4.local: OK
Fri Jan  2 09:59:46 1970 user.notice otbr-agent[2080]: 01:32:55.900 [N] MeshForwarder-: Dropping (reassembly queue) IPv6 UDP msg, len:195, chksum:cfe0, ecn:no, sec:yes, error:ReassemblyTimeout, prio:net, rss:-48.0
Fri Jan  2 09:59:46 1970 user.notice otbr-agent[2080]: 01:32:55.900 [N] MeshForwarder-:     src:[fe80:0:0:0:548e:b5a5:5f0e:f7b2]:19788
Fri Jan  2 09:59:46 1970 user.notice otbr-agent[2080]: 01:32:55.900 [N] MeshForwarder-:     dst:[ff02:0:0:0:0:0:0:1]:19788
Fri Jan  2 09:59:46 1970 user.notice otbr-agent[2080]: 01:32:56.058 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:67, src:9e28a2537926b189, dst:0xffff, tag:1852, offset:128, dglen:195, sec:yes
Fri Jan  2 09:59:46 1970 user.notice otbr-agent[2080]: 01:32:56.261 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:67, src:ae485efcc2d73710, dst:0xffff, tag:13453, offset:128, dglen:195, sec:yes
Fri Jan  2 09:59:47 1970 user.notice otbr-agent[2080]: 01:32:56.863 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:1, src:0xc000, dst:0xb801, tag:7731, offset:504, dglen:505, sec:yes
Fri Jan  2 09:59:47 1970 user.notice otbr-agent[2080]: 01:32:56.897 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:96, src:0xa400, dst:0xb801, tag:59775, offset:408, dglen:505, sec:yes
Fri Jan  2 09:59:47 1970 user.notice otbr-agent[2080]: 01:32:56.949 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:1, src:0xa400, dst:0xb801, tag:59775, offset:504, dglen:505, sec:yes
Fri Jan  2 09:59:49 1970 user.notice otbr-agent[2080]: 01:32:58.906 [N] MeshForwarder-: Dropping (reassembly queue) IPv6 UDP msg, len:505, chksum:0e48, ecn:no, sec:yes, error:ReassemblyTimeout, prio:normal, rss:-62.0
Fri Jan  2 09:59:49 1970 user.notice otbr-agent[2080]: 01:32:58.906 [N] MeshForwarder-:     src:[fde0:a9ae:c6d2:72b8:1ff7:3bf7:8e64:cb64]:49156
Fri Jan  2 09:59:49 1970 user.notice otbr-agent[2080]: 01:32:58.906 [N] MeshForwarder-:     dst:[fde0:a9ae:c6d2:72b8:2219:9550:7e2:30ab]:53548
Fri Jan  2 09:59:49 1970 user.notice otbr-agent[2080]: 01:32:58.906 [N] MeshForwarder-: Dropping (reassembly queue) IPv6 UDP msg, len:505, chksum:a04a, ecn:no, sec:yes, error:ReassemblyTimeout, prio:normal, rss:-62.625
Fri Jan  2 09:59:49 1970 user.notice otbr-agent[2080]: 01:32:58.906 [N] MeshForwarder-:     src:[fde0:a9ae:c6d2:72b8:4abb:e6e7:13d7:443a]:49157
Fri Jan  2 09:59:49 1970 user.notice otbr-agent[2080]: 01:32:58.906 [N] MeshForwarder-:     dst:[fde0:a9ae:c6d2:72b8:2219:9550:7e2:30ab]:53548
Fri Jan  2 09:59:49 1970 user.notice otbr-agent[2080]: 01:32:58.906 [N] MeshForwarder-: Dropping (reassembly queue) IPv6 UDP msg, len:505, chksum:1020, ecn:no, sec:yes, error:ReassemblyTimeout, prio:normal, rss:-63.5
Fri Jan  2 09:59:49 1970 user.notice otbr-agent[2080]: 01:32:58.906 [N] MeshForwarder-:     src:[fde0:a9ae:c6d2:72b8:e67b:4b2e:bd64:a779]:49156
Fri Jan  2 09:59:49 1970 user.notice otbr-agent[2080]: 01:32:58.906 [N] MeshForwarder-:     dst:[fde0:a9ae:c6d2:72b8:2219:9550:7e2:30ab]:53548
Fri Jan  2 09:59:50 1970 user.notice otbr-agent[2080]: 01:32:59.937 [N] MeshForwarder-: Dropping (reassembly queue) IPv6 UDP msg, len:195, chksum:5b22, ecn:no, sec:yes, error:ReassemblyTimeout, prio:net, rss:-48.0
Fri Jan  2 09:59:50 1970 user.notice otbr-agent[2080]: 01:32:59.938 [N] MeshForwarder-:     src:[fe80:0:0:0:548e:b5a5:5f0e:f7b2]:19788
Fri Jan  2 09:59:50 1970 user.notice otbr-agent[2080]: 01:32:59.938 [N] MeshForwarder-:     dst:[ff02:0:0:0:0:0:0:1]:19788
Fri Jan  2 09:59:50 1970 user.notice otbr-agent[2080]: 01:33:00.287 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:1, src:0x1800, dst:0xb801, tag:16229, offset:504, dglen:505, sec:yes
Fri Jan  2 09:59:52 1970 user.notice otbr-agent[2080]: 01:33:01.929 [N] MeshForwarder-: Dropping (reassembly queue) IPv6 UDP msg, len:505, chksum:fe32, ecn:no, sec:yes, error:ReassemblyTimeout, prio:normal, rss:-64.625
Fri Jan  2 09:59:52 1970 user.notice otbr-agent[2080]: 01:33:01.930 [N] MeshForwarder-:     src:[fde0:a9ae:c6d2:72b8:4abb:e6e7:13d7:443a]:49157
Fri Jan  2 09:59:52 1970 user.notice otbr-agent[2080]: 01:33:01.931 [N] MeshForwarder-:     dst:[fde0:a9ae:c6d2:72b8:2219:9550:7e2:30ab]:53548
Fri Jan  2 09:59:54 1970 user.notice otbr-agent[2080]: 01:33:03.912 [N] MeshForwarder-: Dropping (reassembly queue) IPv6 UDP msg, len:505, chksum:9089, ecn:no, sec:yes, error:ReassemblyTimeout, prio:normal, rss:-63.25
Fri Jan  2 09:59:54 1970 user.notice otbr-agent[2080]: 01:33:03.912 [N] MeshForwarder-:     src:[fde0:a9ae:c6d2:72b8:fbf5:9ee7:d441:2f1e]:49157
Fri Jan  2 09:59:54 1970 user.notice otbr-agent[2080]: 01:33:03.912 [N] MeshForwarder-:     dst:[fde0:a9ae:c6d2:72b8:2219:9550:7e2:30ab]:53548
Fri Jan  2 09:59:54 1970 user.notice otbr-agent[2080]: 01:33:04.234 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:96, src:0x1400, dst:0xb801, tag:16108, offset:120, dglen:505, sec:yes
Fri Jan  2 09:59:54 1970 user.notice otbr-agent[2080]: 01:33:04.272 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:96, src:0xa400, dst:0xb801, tag:59778, offset:120, dglen:505, sec:yes
Fri Jan  2 09:59:54 1970 user.notice otbr-agent[2080]: 01:33:04.291 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:96, src:0xa400, dst:0xb801, tag:59778, offset:216, dglen:505, sec:yes
Fri Jan  2 09:59:54 1970 user.notice otbr-agent[2080]: 01:33:04.323 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:96, src:0xa400, dst:0xb801, tag:59778, offset:312, dglen:505, sec:yes
Fri Jan  2 09:59:54 1970 user.notice otbr-agent[2080]: 01:33:04.362 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:96, src:0xa400, dst:0xb801, tag:59778, offset:408, dglen:505, sec:yes
Fri Jan  2 09:59:54 1970 user.notice otbr-agent[2080]: 01:33:04.366 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:1, src:0xa400, dst:0xb801, tag:59778, offset:504, dglen:505, sec:yes
Fri Jan  2 09:59:54 1970 user.notice otbr-agent[2080]: 01:33:04.532 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:96, src:0x1400, dst:0xb801, tag:16108, offset:216, dglen:505, sec:yes
Fri Jan  2 09:59:54 1970 user.warn otbr-agent[2080]: [WARN]-MDNS----: Failed to send goodbye message for host 2E99AC88EB37D548.local: OK
Fri Jan  2 09:59:55 1970 user.notice otbr-agent[2080]: 01:33:04.954 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:96, src:0x1400, dst:0xb801, tag:16108, offset:312, dglen:505, sec:yes
Fri Jan  2 09:59:55 1970 user.notice otbr-agent[2080]: 01:33:04.994 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:96, src:0x1400, dst:0xb801, tag:16108, offset:408, dglen:505, sec:yes
Fri Jan  2 09:59:55 1970 user.notice otbr-agent[2080]: 01:33:05.021 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:1, src:0x1400, dst:0xb801, tag:16108, offset:504, dglen:505, sec:yes
Fri Jan  2 09:59:55 1970 user.notice otbr-agent[2080]: 01:33:05.785 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:96, src:0xb400, dst:0xb801, tag:65141, offset:496, dglen:818, sec:yes
[... cut ...]
Fri Jan  2 09:59:55 1970 user.notice otbr-agent[2080]: 01:33:05.800 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:88, src:0x0402, dst:0xb801, tag:20857, offset:200, dglen:505, sec:yes
Fri Jan  2 09:59:55 1970 user.notice otbr-agent[2080]: 01:33:05.820 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:34, src:0xb400, dst:0xb801, tag:65141, offset:784, dglen:818, sec:yes
Fri Jan  2 09:59:56 1970 user.notice otbr-agent[2080]: 01:33:05.867 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:88, src:0x0402, dst:0xb801, tag:20857, offset:288, dglen:505, sec:yes

luyou-tuya, 2022-08-26:
> @luyou-tuya , can you provide more information on your RCP serial connection at the OTBR? Are you using UART or SPI? What baud rate? We have seen some cases where the serial transfer rate between RCP and host is slower than what is supported by the 802.15.4 radio itself.
Thank you for your reply! We are using a serial port, the baud rate of the serial port is 460800, and the hardware flow control is turned on.

luyou-tuya, 2022-08-26:
> 

I have a question that after otbr restarts, it becomes a child, what should I do so that it does not become a child, but becomes a leader or route

luyou-tuya, 2022-08-26:
> @luyou-tuya , can you provide more information on your RCP serial connection at the OTBR? Are you using UART or SPI? What baud rate? We have seen some cases where the serial transfer rate between RCP and host is slower than what is supported by the 802.15.4 radio itself.

I have a question that after otbr restarts, it becomes a child, what should I do so that it does not become a child, but becomes a leader or route

jwhui, 2022-08-26:
> Thank you for your reply! We are using a serial port, the baud rate of the serial port is 460800, and the hardware flow control is turned on.

From the logs, it does not seem like the host is receiving all that many frames (i.e. less than 10 frames per second). Have you done any throughput testing using ping or iperf between two nodes? I raise this since we've seen issues in the past where the data rate between RCP and host is lower than expected.

jwhui, 2022-08-26:
> I have a question that after otbr restarts, it becomes a child, what should I do so that it does not become a child, but becomes a leader or route

How long does it take for OTBR to restart? A Thread device that was previously Leader or Router will attempt to send a multicast MLE Link Request to reestablish its previous role. However, this only works if there are neighboring Leader/Router devices that can verify its prior role. If OTBR takes too long to restart, then it could be that the neighboring devices have already timed out state for the device.

If the device is not able to get a response to its multicast MLE Link Request, it will reattach as if it was joining a new partition. Thread requires all devices joining a new partition to join as a child first.

jwhui, 2022-09-23:
Closing stale issue.

closed 2022-09-23 by jwhui