openthread/openthread #8089: Can the ot-ctl command be called concurrently?

lancersky, 2022-08-26:
**Describe the bug** A clear and concise description of what the bug is.
I opened five terminals to send the `ot-ctl` command as follows.
```
for i in $(seq 1 2000)
do
	ot-ctl state
	echo "count" $i
	sleep 1
done
```
[image]
And I get an error `read: Connection reset by peer` in every terminal.
It may not get the correct response in a concurrent situation. 
**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id https://github.com/openthread/ot-br-posix/commit/3a98779dc9f0875af9c7b137e764c9663bfdc9c3
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

simonlingoogle, 2022-08-26:
The short answer is that ot-ctl does not support concurrent executions.

lancersky, 2022-08-28:
thanks for your reply.

closed 2022-08-28 by lancersky