openthread/openthread #8101: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs

lancersky, 2022-08-30:
**Describe the bug** A clear and concise description of what the bug is.
We deployed 100 devices with different PSKd. All devices are commissioning at the same time.

```
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.715 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.719 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.724 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.729 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.735 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.741 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.748 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.754 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.759 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.762 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.763 [W] Platform------: [netif] Failed to transmit, error:NoBufs
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.768 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.772 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.779 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.notice otbr-agent[16861]: 00:29:28.788 [N] MeshForwarder-: Dropping rx frag frame, error:NoBufs, len:90, src:6218910bf1c079d1, dst:0xffff, tag:63840, offset:0, dglen:167, sec:yes
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.791 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.notice otbr-agent[16861]: 00:29:28.792 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:39, src:6218910bf1c079d1, dst:0xffff, tag:63840, offset:128, dglen:167, sec:yes
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.795 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.800 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.805 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.812 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.notice otbr-agent[16861]: 00:29:28.815 [N] MeshForwarder-: Dropping rx frag frame, error:NoBufs, len:90, src:6218910bf1c079d1, dst:0xffff, tag:63841, offset:0, dglen:167, sec:yes
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.818 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.notice otbr-agent[16861]: 00:29:28.819 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:39, src:6218910bf1c079d1, dst:0xffff, tag:63841, offset:128, dglen:167, sec:yes
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.823 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.827 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.832 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.842 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.848 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
Tue Aug 23 18:29:33 2022 user.warn otbr-agent[16861]: 00:29:28.853 [W] MeshCoP-------: Failed to send MGMT_COMMISSIONER_SET.req: NoBufs
```
The OpenThread configuration is as follows
```
	-DOPENTHREAD_CONFIG_COMMISSIONER_MAX_JOINER_ENTRIES=100 \
	-DOPENTHREAD_CONFIG_PLATFORM_RADIO_SPINEL_RX_FRAME_BUFFER_SIZE=16384 \
	-DOPENTHREAD_CONFIG_NUM_MESSAGE_BUFFERS=16383 \
	-DOPENTHREAD_CONFIG_MLE_MAX_CHILDREN=100 \
	-DOPENTHREAD_CONFIG_TMF_ADDRESS_CACHE_ENTRIES=200 \
```
How can we optimize?
**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id  https://github.com/openthread/ot-br-posix/commit/3a98779dc9f0875af9c7b137e764c9663bfdc9c3
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

jwhui, 2022-08-30:
@lancersky , thanks for raising this issue.

Can you provide more insight into how you are driving the Commissioner? A `MGMT_COMMISSIONER_SET.req` message is sent each time a joiner is added to the Commissioner. I suspect the `NoBufs` error is due to issuing multiple `joiner add` commands back-to-back. The OpenThread Commissioner is relatively simple and does not have any throttling mechanism when the user requests to add joiners back-to-back quickly. If this is the issue, is it possible to add some delay between each joiner add?

lancersky, 2022-08-30:
Hi @jwhui ,
I drove the Commissioner with the following command and no delay between each `joiner add`. I will add some delay to have a try. Thanks.
```
ubus call otbr joineradd '{"pskd":"String","eui64":"String","timeout":"Integer"}'
```

lancersky, 2022-08-30:
Hi @jwhui 
I have another question. 
Do I need to execute `commissioner joiner remove` before `commissioner stop`?

jwhui, 2022-08-30:
> Hi @jwhui I have another question. Do I need to execute `commissioner joiner remove` before `commissioner stop`?

No, calling stop will also clear the joiner list.

https://github.com/openthread/openthread/blob/a6032491b85e49b2c0a42ffd9ba6be4205bddcb7/src/core/meshcop/commissioner.cpp#L348

jwhui, 2022-10-20:
Closing stale issue.

closed 2022-10-20 by jwhui