openthread/openthread #8117: openthread/src/core/common/message.hpp:122:38: error: ‘nonnull’ argument ‘aMessage’ compared to NULL [-Werror=nonnull-compare]

zyga, 2022-09-02:
**Describe the bug** A clear and concise description of what the bug is.

Build failure with gcc 12.2.0

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id: 30d65bca4896a38e0e8a9e0b3cb0b9d458454ef2
2. IEEE 802.15.4 hardware platform: N/A
3. Build steps: `./bootstrap && ./script/cmake-build posix -DOT_SPINEL_RESET_CONNECTION=ON -DOT_THREAD_VERSION=1.2 -DOT_DAEMON=ON`

4. Network topology: N/A

**Expected behavior** A clear and concise description of what you expected to happen.

Build completes without warnings or errors

**Console/log output** If applicable, add console/log output to help explain your problem.

```
In file included from /home/zyga/projects/openthread/src/core/crypto/aes_ccm.hpp:43,
                 from /home/zyga/projects/openthread/src/core/thread/mle.hpp:45,
                 from /home/zyga/projects/openthread/src/core/thread/mle.cpp:34:
/home/zyga/projects/openthread/src/core/thread/mle.cpp: In member function ‘void ot::Mle::Mle::SendDelayedResponse(TxMessage&, const DelayedResponseMetadata&)’:
/home/zyga/projects/openthread/src/core/common/message.hpp:122:38: error: ‘nonnull’ argument ‘aMessage’ compared to NULL [-Werror=nonnull-compare]
  122 |         if (((aError) != kErrorNone) && ((aMessage) != nullptr)) \
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/zyga/projects/openthread/src/core/thread/mle.cpp:1643:5: note: in expansion of macro ‘FreeMessageOnError’
 1643 |     FreeMessageOnError(&aMessage, error);
      |     ^~~~~~~~~~~~~~~~~~
```

**Additional context** Add any other context about the problem here.

```
zyga@lambert:~/projects/openthread$ gcc --version
gcc (Ubuntu 12.2.0-1ubuntu1) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

zyga@lambert:~/projects/openthread$ cmake --version
cmake version 3.24.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

jwhui, 2022-09-02:
@zyga , thanks for raising this issue. I just submitted #8118. Please review.

zyga, 2022-09-02:
@jwhui looks great, thank you!

I was wondering about the macro, why is that used? If this was a function the error would have been easier to fix and performance would be identical, given current compilers.

Still, I'm mostly curious as I'm just learning about Open Thread, perhaps the utility of the macro eludes me.

zyga, 2022-09-02:
@abtink thank you for the exhaustive explanation!

PR #8118 (2022-09-06): [github-actions] add gcc-12 to build matrix and fix warnings
Fixes #8117
closed 2022-09-06 by jwhui