openthread/openthread #8172: [github-actions] Cert_5_3_05_RoutingLinkQuality.py fails intermittently

jwhui, 2022-09-15:
```
======================================================================
FAIL: test (__main__.Cert_5_3_5_RoutingLinkQuality)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/runner/work/openthread/openthread/examples/../tests/scripts/thread-cert/Cert_5_3_05_RoutingLinkQuality.py", line 104, in test
    self.assertTrue(self.nodes[ROUTER3].ping(leader_rloc))
AssertionError: False is not true

----------------------------------------------------------------------
```
https://github.com/openthread/openthread/actions/runs/3061813605/jobs/4942052869

abtink, 2022-09-15:
+1 I have also noticed this test failing (similar error) more frequently lately.

jwhui, 2022-10-11:
https://github.com/openthread/openthread/actions/runs/3228459474/jobs/5284585551

jwhui, 2022-10-18:
https://github.com/openthread/openthread/actions/runs/3277136161/jobs/5394028879

closed 2024-06-21 by jwhui