openthread/openthread #8179: [mle] no delay between MGMT SET requests

edmont, 2022-09-16:
Thread 1.1 Certification test 9.2.9 is failing for the Leader DUT on step 24 when an OpenThread based reference device is being used as Router_1 in the topology.

This is due to the fact that, when processing Child Id Response, Router_1 notices the Leader has both Active and Pending Datasets outdated and sends (`DatasetManager::HandleDatasetUpdated()`) MGMT_ACTIVE_SET.req and MGMT_PENDING_SET.req almost without delay (10-12 ms), leaving no room for the Leader to send MGMT_ACTIVE_SET.rsp, MGMT_PENDING_SET.rsp, and MGMT_DATASET_CHANGED.ntf.

https://github.com/openthread/openthread/blob/main/src/core/thread/mle.cpp#L2921-L2941
```
        // Active Dataset
        if (hasActiveTimestamp)
        {
            if (activeDatasetOffset > 0)
            {
                IgnoreError(aRxInfo.mMessage.Read(activeDatasetOffset, tlv));
                IgnoreError(Get<MeshCoP::ActiveDatasetManager>().Save(
                    activeTimestamp, aRxInfo.mMessage, activeDatasetOffset + sizeof(tlv), tlv.GetLength()));
            }
        }

        // Pending Dataset
        if (hasPendingTimestamp)
        {
            if (pendingDatasetOffset > 0)
            {
                IgnoreError(aRxInfo.mMessage.Read(pendingDatasetOffset, tlv));
                IgnoreError(Get<MeshCoP::PendingDatasetManager>().Save(
                    pendingTimestamp, aRxInfo.mMessage, pendingDatasetOffset + sizeof(tlv), tlv.GetLength()));
            }
        }
```

The test is passing when using Kirale device as Router_1 since it allows for around 200 ms between both MGMT SET requests.

Similar delay should be implemented in OpenThread as well.

simonlingoogle, 2022-09-16:
The issue is already fixed in https://github.com/openthread/openthread/pull/6813

But the 1.1 golden device is using 202008 reference release, which does not have this fix.

simonlingoogle, 2022-09-16:
Creating a new OT 1.1 reference release using latest OpenThread code can fix this issue, but it may have unexpected impact on the Thread 1.1 certification. 

I would suggest to create a branch based on 202008 reference release, and fix the issue on that branch.

Thoughts? @edmont @librasungirl @jwhui

edmont, 2022-09-16:
> I would suggest to create a branch based on 202008 reference release, and fix the issue on that branch.

SGTM.

edmont, 2022-10-03:
@jwhui any thoughts about @simonlingoogle's proposition?

jwhui, 2026-05-09:
Closing stale issue.