openthread/openthread #8214: otbr-agent segment error

lancersky, 2022-09-29:
**Describe the bug** A clear and concise description of what the bug is.
We test whether Thread can work properly in the case of the wifi throughput test.
The OTBR is a single node and continuously executes `ot-ctl discover` command with 1 second delay.
[otbr-agent segment error 3.txt](https://github.com/openthread/openthread/files/9672437/otbr-agent.segment.error.3.txt)
[otbr-agent segment error 1.txt](https://github.com/openthread/openthread/files/9672439/otbr-agent.segment.error.1.txt)
[otbr-agent segment error 2.txt](https://github.com/openthread/openthread/files/9672440/otbr-agent.segment.error.2.txt)

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id
https://github.com/openthread/ot-br-posix/commit/3a98779dc9f0875af9c7b137e764c9663bfdc9c3
The Thread RCP firmware built from GSDK 4.1.0
3. IEEE 802.15.4 hardware platform
Silicon Labs EFR32MG21F768
4. Build steps
5. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

jwhui, 2022-09-29:
@lancersky , this appears to be a duplicate of #8110 , which should have been resolved with #8111. Can you move to a later commit that includes this fix and see if that helps?

lancersky, 2022-10-09:
@jwhui.Thanks, It work.

closed 2022-10-09 by lancersky