openthread/openthread #8223: SRP client is not selecting the correct SRP server out of multiple

EskoDijk, 2022-09-30:
**Describe the bug** A clear and concise description of what the bug is.
If an SRP client has registered its service(s) to an SRP server already, and a new SRP server is added with a numerically lower IPv6 address, the SRP client does not seem to (re)register its services to the new SRP server.

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id: ccf8ea69230bf474cbcd61ef61fc5519df7665dd (Sep 6)
2. IEEE 802.15.4 hardware platform: otns simulation
3. Build steps: regular build with SRP server and client 'on'
4. Network topology: 2 routers, connected

Steps are
* create topology; node 2 has lower IPv6 address number
* enable srp server on node 1
* node 1 and/or 2: register a service with SRP
* enable srp server on node 2
* services should be registered to node 2 (but it remains empty)

**Expected behavior** A clear and concise description of what you expected to happen.
SRP client reregisters to new SRP server with numerically lower address.

**Console/log output** If applicable, add console/log output to help explain your problem.
can add if needed

**Additional context** Add any other context about the problem here.
Thread spec 1.3.0 ; 14.3.5.1 defines the requirements.

abtink, 2022-09-30:
Thanks @EskoDijk. I checked the implementation and I believe this is intended behavior.

- Regarding unicast entries in network data, the implemented behavior is that client will stay with its previously selected server as long it is seen/present in the Network Data. 
- So even if a new preferred unicast entry is seen later, we stay with our previously selected server. 
- We also have a build feature `OPENTHREAD_CONFIG_SRP_CLIENT_SAVE_SELECTED_SERVER_ENABLE` which retains the selected server info in non-volatile and tries to re-use it on a reboot.  

Here is the related code when we try to select a unicast entry (of certain origin whether info is in Server data or Service Data):
```c++
Error Client::SelectUnicastEntry(DnsSrpUnicast::Origin aOrigin, DnsSrpUnicast::Info &aInfo) const
{
    ...

    while (Get<NetworkData::Service::Manager>().GetNextDnsSrpUnicastInfo(iterator, unicastInfo) == kErrorNone)
    {
        if (unicastInfo.mOrigin != aOrigin)
        {
            continue;
        }

        if (mAutoStart.HasSelectedServer() && (GetServerAddress() == unicastInfo.mSockAddr))
        {
            aInfo = unicastInfo;
            error = kErrorNone;
            ExitNow();
        }
```

------

The reason for this model is to avoid conflict when user adds new BRs in the set-up:
- Suppose  we have a have BR and a bunch of nodes/client already registered their services with it.
- The BR acting as adv proxy, will be advertising and claiming all names on AIL.
- Now user install a new BR which happen to have a smaller ML IPv6 address and publishes its info in net data.
- If we now have all clients switch to new server, they will all get name conlfict error sicne the previous server is still adv them
   - I expect this situation to resolve itself eventually but can mess stuff up for a while
   - Clients will try a bunch of time with the new server, fail and eventually try the next sever and end up with their original server (OT code does add this behavior under `OPENTHREAD_CONFIG_SRP_CLIENT_SWITCH_SERVER_ON_FAILURE`).
   - Or if we don't have retry/switch mechanism on client side, we need to wait for service to age and timeout on original server before client can register with new server.
 - Overall I think the implemented behavior make sense (if I recall correctly it came up during earlier testing). I don't see any immediate benefit for us to have client to switch in such a case.
 - Thoughts?

Couple of things to point:
- We have different types of entries in Network Data:
     1) Preferred unicast entries with address included in service data.
     2) Anycast entries (each having a seq number).
     3) Unicast entries with address info included in server data.
- On clinet if we see discover a server using different and more preferred entry type, we do switch to that server
   - For example, if we are currently using a unicast entry server, and discover a new anycast entry we switch. 
- Also if we are using anycast server and see new entry with a preferred seq num, we switch.
- Only if we have previously selected a unicast entry and we only see entries on the same type, we try to stay with our previously selected server.

EskoDijk, 2022-10-03:
Thanks @abtink , that clarifies the behavior. I've created issue SPEC-1113 for further spec discussion. 
In this case I've manually added a 2nd SRP server (with unicast dataset). In automatic SRP server scenarios, it may be beneficial if only one SRP Server is advertised by Unicast Dataset - i.e. one could withdraw its automatically added service.

That could be based on numerically lowest IPv6 address, but it may also play a role how well connected the device offering the SRP server is. E.g. a Border Router with backbone link is a better candidate to host it than an on-mesh only device (e.g. BR without external connectivity)

Finally a more generic solution would be for the client to first unregister everything on its existing server and then register again with the newly preferred server.

abtink, 2022-10-03:
> I've created issue SPEC-1113 for further spec discussion.

Thanks. 👍 

> In this case I've manually added a 2nd SRP server (with unicast dataset). In automatic SRP server scenarios, it may be beneficial if only one SRP Server is advertised by Unicast Dataset - i.e. one [...]

We can consider this (have servers backoff and remove their entry from netdata if they see other server), but at the same time, it may be good to keep some for redundancy.

> Finally a more generic solution would be for the client to first unregister everything on its existing server and then register again with the newly preferred server.

We considered this, but it adds complexity and also some potential temporary disruption (no immediate benefit).
- When we unregister with the first server, it will advertise this on mDNS side and may cause devices on AIL to take action (thinking certain nodes/services are removed). 
-  This model  also may not work if the client is rebooted (not get the chance to unregister with first server).

From the client perspective when using  "unicast" server address, if the client is happy with a currently selected server, there does not seem to be any advantage for it to switch (all "unicast" servers are similar from client prespective). 
- Earlier in client code we let it choose one randomly among all unicast entries with idea to balance the load and traffic when we have multiple servers.

EskoDijk, 2022-10-03:
Ok agree you have defined a good solution here.

The remaining open issue we can take to SPEC-1113: the case when e.g. one SRP server is on a BR with backbone link connection (and has advertising proxy etc active) while another SRP server is on a BR without backbone link. 

>  there does not seem to be any advantage for it to switch

For the above case, if the client has registered to the non-connected BR initially, and later a backbone-connected BR joins, there's a great advantage for the client to switch - its service being able to be found e.g. by Ethernet or Wi-Fi devices.

jwhui, 2022-10-20:
> For the above case, if the client has registered to the non-connected BR initially, and later a backbone-connected BR joins, there's a great advantage for the client to switch - its service being able to be found e.g. by Ethernet or Wi-Fi devices.

In this case, I think the SRP server that is not connected to an infrastructure link should stop publishing its SRP serve when it detects that there is an SRP server with an attached infrastructure link.

EskoDijk, 2022-10-21:
One other interesting thing to note is that the feature of switching to the next SRP server, in case the first one reports back an error in the registration, is fine per the IETF SRP-draft but clashes with the Thread specified behavior of picking the numerically lowest SRP server. Also noted this in SPEC-1113.

jwhui, 2022-12-07:
Closing stale issue.

closed 2022-12-07 by jwhui