openthread/openthread #8256: missing leader node info in diagnostics restful api

lancersky, 2022-10-08:
**Describe the bug** A clear and concise description of what the bug is.
[image]
[image]

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id https://github.com/openthread/ot-br-posix/commit/3a98779dc9f0875af9c7b137e764c9663bfdc9c3
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

lancersky, 2022-10-09:
There are 16 router devices and 65 child devices in the Thread network.

lancersky, 2022-10-27:
Hi @jwhui 
Is it related to the number of TlvTypes? More TlvTypes take more time to collect?
[image]

martinzi, 2022-11-03:
thanks for raising this discussion.

For further tracing down your issue, I propose comparing wireshark captures. Typically, collect timeout of two seconds is sufficient for multicast responses from routers. However I suspect you may occasionally see packet collissions, in particular related to spacially dispersed Thread routers. Currently, OT lacks randomly delayed answers for multicast management queries. This is addressed in [SPEC-1061](https://threadgroup.atlassian.net/browse/SPEC-1061)

jwhui, 2023-01-26:
Closing stale issue.

closed 2023-01-26 by jwhui