openthread/openthread #8281: TCP CLI module will crash due to multiple incoming connections when listening

dsyx, 2022-10-14:
**Describe the bug**

Refer to [OpenThread CLI - TCP Example](https://github.com/openthread/openthread/blob/main/src/cli/README_TCP.md).
Enable TCP listening on Node A, Node B connects to A, then Node B disconnects. Then node C connects to A, and A's shell will freeze.

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id `ba006225e03c57aeb09599b5acb67e09eb98ee95`
2. IEEE 802.15.4 hardware platform `efr32`

**Expected behavior**

Node A's shell should work fine.

**Console/log output**

Node A:

```bash
> tcp init
Done
> tcp listen :: 10099
Done
> Accepted connection from [240e:3b6:4a7:4380:489e:d1f7:404e:fb8c]:42720
TCP: Connection established
TCP: Reached end of stream    # Node B is disconnected
# At this point, node C is connected to node A. A's shell freezes and cannot be typed.
```

samkumar, 2022-10-16:
Thanks @dsyx for identifying this issue. I was able to reproduce it and I've proposed a fix in Pull Request #8291.

In your example, note that the connection from A-B is only _half-closed_ at the time that C connects. As far as I can tell the bug only happens after B has run `tcp sendend` but before A has run `tcp sendend`. Thus, when C connects, the A's TCP endpoint is still busy with the connection to B.

The intended behavior of the TCP CLI tool, in this case, is to silently drop the incoming connection requests from C and print a warning message. B's connection request will be retransmitted, so the warning message at A will be printed once for each such transmission attempt. The warnings will stop once C hits the maximum number of retransmissions and times out, or once A tears down the connection with B by running `tcp sendend`, at which point C's next connection attempt will succeed.

Here's example output at A, with the bug fixed:
```
> tcp init circular
Done
> tcp listen :: 40000
Done
> ipaddr
fd14:3789:3629:7fd3:0:ff:fe00:fc00
fd14:3789:3629:7fd3:0:ff:fe00:d000
fd14:3789:3629:7fd3:4988:3d87:a565:3415
fe80:0:0:0:dc19:3dce:a71b:5ef4
Done
> Accepted connection from [fd14:3789:3629:7fd3:89b9:f2b:7bf1:9d83]:49152
TCP: Connection established
TCP: Reached end of stream
TCP: Ignoring incoming connection request from [fd14:3789:3629:7fd3:4988:3d87:a565:3415]:49152 (active socket is busy)
TCP: Ignoring incoming connection request from [fd14:3789:3629:7fd3:4988:3d87:a565:3415]:49152 (active socket is busy)
TCP: Ignoring incoming connection request from [fd14:3789:3629:7fd3:4988:3d87:a565:3415]:49152 (active socket is busy)
TCP: Ignoring incoming connection request from [fd14:3789:3629:7fd3:4988:3d87:a565:3415]:49152 (active socket is busy)
TCP: Ignoring incoming connection request from [fd14:3789:3629:7fd3:4988:3d87:a565:3415]:49152 (active socket is busy)
tcp sendend
Done
> TCP: Disconnected
Accepted connection from [fd14:3789:3629:7fd3:cb17:89dd:4f6a:207]:49152
TCP: Connection established
```
Note that the `tcp sendend` line is something that I typed at the terminal. Doing this tore down the connection with B, allowing C's next connection attempt to succeed (evidenced by the "Accepted connection from ..." line).

samkumar, 2022-10-16:
To provide a bit more context: the TCP API allows an application to deal with an incoming TCP connection request in three ways: (1) accepting the connection, (2) silently ignoring the request, or (3) refusing the request (i.e., send an RST segment to the requesting node so that it knows not to retry). The current behavior chosen by the CLI tool is the second choice, which is why we see C continue to retry its connection establishment request. The advantage is that, if the TCP endpoint becomes free, the connection will eventually be established.

With a one-line change to the CLI tool, I could choose the third choice instead, which would cause C to not retry. I've currently chosen the second choice, but I'm open to feedback if others find the third choice more intuitive.

dsyx, 2022-10-17:
@samkumar 
Actually, I made some modifications to the source code of TCP CLI::

```cpp
void TcpExample::HandleTcpEstablished(otTcpEndpoint *aEndpoint)
{
    OT_UNUSED_VARIABLE(aEndpoint);
    OT_ASSERT(aEndpoint == &mEndpoint);

    OutputLine("TCP: Connection established");
    mEndpointConnected = true;
}
```

```cpp
otTcpIncomingConnectionAction TcpExample::HandleTcpAcceptReady(otTcpListener *   aListener,
                                                               const otSockAddr *aPeer,
                                                               otTcpEndpoint **  aAcceptInto)
{
    OT_UNUSED_VARIABLE(aListener);

    if (mEndpointConnected)
    {
        OutputFormat("TCP: Ignoring incoming connection request from ");
        OutputSockAddr(*aPeer);
        OutputLine(" (active socket is busy)");

        return OT_TCP_INCOMING_CONNECTION_ACTION_REFUSE;
    }

    *aAcceptInto = &mEndpoint;
    return OT_TCP_INCOMING_CONNECTION_ACTION_ACCEPT;
}
```

I roughly traced the call to `HandleTcpAcceptReady`. It is called by `tcplp_sys_accept_ready` in `tcp6.cpp` and returns `(struct tcpcb *)-1`:

```cpp
    switch (action)
    {
    case OT_TCP_INCOMING_CONNECTION_ACTION_ACCEPT:
    {
        Tcp::Endpoint &endpoint = AsCoreType(endpointPtr);

        /*
         * The documentation says that the user must initialize the
         * endpoint before passing it here, so we do a sanity check to make
         * sure the endpoint is initialized and closed. That check may not
         * be necessary, but we do it anyway.
         */
        VerifyOrExit(tcp.IsInitialized(endpoint) && endpoint.IsClosed());

        rv = &endpoint.GetTcb();

        break;
    }
    case OT_TCP_INCOMING_CONNECTION_ACTION_DEFER:
        rv = nullptr;
        break;
    case OT_TCP_INCOMING_CONNECTION_ACTION_REFUSE:
        rv = (struct tcpcb *)-1;
        break;
    }

exit:
    return rv;
```

`tcp_input` in `tcp_input.c` calls `tcplp_sys_accept_ready` and jumps to `dropwithreset` based on the return value:

```cpp
tp = tcplp_sys_accept_ready(tpl, &ip6->ip6_dst, th->th_sport); // Try to allocate an active socket to accept into
if (tp == NULL) {
    /* If we couldn't allocate, just ignore the SYN. */
    return IPPROTO_DONE;
}
if (tp == (struct tcpcb *) -1) {
    rstreason = ECONNREFUSED;
    goto dropwithreset;
}
```

However `tcp_dropwithreset` doesn't seem to handle `(struct tcpcb *)-1` specially:

```cpp
dropwithreset:

    /*
     * samkumar: The check against inp != NULL is now a check on tp != NULL.
     */
    if (tp != NULL) {
        tcp_dropwithreset(ip6, th, tp, tp->instance, tlen, rstreason);
    } else
        tcp_dropwithreset(ip6, th, NULL, tpl->instance, tlen, rstreason);
    goto drop;

dropunlock:
drop:
    return (IPPROTO_DONE);
```

This can lead to memory access errors.

dsyx, 2022-10-17:
According to #8291, I made the same modification to `tcp_input.c`:

```cpp
tp = tcplp_sys_accept_ready(tpl, &ip6->ip6_dst, th->th_sport); // Try to allocate an active socket to accept into
if (tp == NULL) {
    /* If we couldn't allocate, just ignore the SYN. */
    return IPPROTO_DONE;
}
if (tp == (struct tcpcb *) -1) {
    rstreason = ECONNREFUSED;
    tp = NULL;
    goto dropwithreset;
}
```

The program behaves as expected.

However, `tcplp_sys_accept_ready` has the possibility to return `(struct tcpcb *)-1`, which may make it a trap in the future.

samkumar, 2022-10-17:
Hi @dsyx,
Your observations about the bad memory access stemming from leaving `tp` as `(struct tcpcb *) -1` are correct. I think that Pull Request #8291 already fixes this. It sounds like the fix also caused the program to exhibit to expected behavior on your end. Thanks for trying it out.

You shouldn't need to modify the CLI tool after Pull Request #8291 (note that it returns `OT_TCP_INCOMING_CONNECTION_ACTION_DEFER` rather than `OT_TCP_INCOMING_CONNECTION_ACTION_REFUSE`).

dsyx, 2022-10-17:
Thanks, @samkumar

PR #8291 (2022-10-17): [tcplp, cli-tcp] fix illegal memory access when refusing incoming TCP connection
This pull request fixes a bug in TCPlp that causes an illegal memory access when refusing an incoming TCP connection using the `OT_TCP_INCOMING_CONNECTION_ACTION_REFUSE` action in the `otTcpAcceptReady` callback. It also fixes a bug in the TCP CLI tool that causes it to attempt to accept every incoming TCP connection (which should not normally be a problem, but in this case was triggering the above bug). Since there are really two separate bugs here, I separated their fixes into separate commits.

This fixes #8281 [... cut]
closed 2022-10-17 by dsyx