openthread/openthread #8295: [srp-client] No API for reliable service update followed by service removal

Damian-Nordic, 2022-10-17:
**Describe the bug**
Currently, the service update can be performed as a combination of two steps:
1. _Clear service_ which makes the SRP client forget the service without sending an SRP update.
2. _Add service_ which re-adds the updated service.

It causes a problem if such an update procedure is followed immediately by removal of the service because:
a. SRP updates are deferred/sent 10-700ms after making a change on the SRP client side.
b. Removal of a service that hasn't been sent to the SRP server yet is done only locally due to the following optimization:
```c++
    if (aService.GetState() == kToAdd)
    {
        // If the service has not been added yet, we can remove it immediately.
        aService.SetState(kRemoved);
    }
```

Consequenty, if an update and removal of the service happens within 700ms or less, it may happen that an SRP update that removes the service will never be emitted though the SRP client will remove the service from its list.

**To Reproduce**

1. Build and flash CLI sample for nRF52840 DK
2. Run OTBR and form a network using default settings
3. Run the following commands on the CLI sample:
```
ifconfig up
dataset set active 000300000e02081111111122222222051000112233445566778899aabbccddeeff01021234
srp client autostart enable
srp client host name testhost
srp client host address auto
srp client callback enable
thread start
srp client service add testinst _test._udp 12345 0 0 054b45593d30
```
4. Verify on the OTBR ot-cli that the testinst service has been registered successfully.
5. Paste the following commands to the UART console of the CLI sample (all at once):
```
srp client service clear testinst _test._udp
srp client service add testinst _test._udp 12345 0 0 054b45593d31
srp client service remove testinst _test._udp
```
6. Observe that the service has been removed from the SRP client:
```
$ srp client service
Done
```
7. However, the service is still registered at the SRP server:
```
> srp server service
testinst._test._udp.default.service.arpa.
    deleted: false
    subtypes: (null)
    port: 12345
    priority: 0
    weight: 0
    ttl: 7200
    lease: 7200
    key-lease: 680400
    TXT: [KEY=30]
    host: testhost.default.service.arpa.
    addresses: [fd11:55:0:0:c3fb:dc9c:54c5:1614]
```

**Expected behavior**
`srp server service` command run on the OTBR ot-ctl returns an empty list.

Perhaps, `otSrpClientRemoveService` should always send an SRP update as it is quite unlikely that a user often adds a new service and removes it just after a couple of milliseconds so the optimization may not be actually useful. Alternatively, it should be possible to re-add the same service using `otSrpClientAddService`, which would use a state, such as `OT_SRP_CLIENT_ITEM_STATE_TO_REFRESH` to indicate that the service needs to be updated.

**Console/log output**

**Additional context**
The issue has been found in our internal stress tests of Matter which remove a DUT from a Matter fabric and commission it again many times in a loop. Not sure how likely it is to occur in real life, but anyway, would be good to have a method to reliably remove the service in each situation.

abtink, 2022-10-17:
Thank @Damian-Nordic for reporting and finding this.

> Perhaps, `otSrpClientRemoveService()` should always send an SRP update as it is quite unlikely that a user often adds a new service and removes it just after a couple of milliseconds so the optimization may not be actually useful.

I agree with you on this and I think this may be the best way. 

Other idea is to add a param to `otSrpClientRemoveService()` to allow user to specify whether they want it to be always unregistered/sent with server. We have something similar with `otSrpClientRemoveHostAndServices()` and `aSendUnregToServer` but the downside I see with this is that it requires a change to public API and impact existing code using the API.

I think simply removing the optimization (always unregister) may be the best way. I can submit a PR changing this.

abtink, 2022-10-18:
Submitted 

Did a quick test (with example above) to validate:
```bash
> srp client service
instance:"testinst", name:"_test._udp", state:Registered, port:12345, priority:0, weight:0
Done

> srp client service clear testinst _test._udp
00:01:41.690 [I] SrpClient-----: Service Registered -> Removed, "testinst" "_test._udp"
Done
> srp client service add testinst _test._udp 12345 0 0 054b45593d31
00:01:41.690 [I] SrpClient-----: Service Removed -> ToAdd, "testinst" "_test._udp"
00:01:41.690 [I] SrpClient-----: subtypes:[] port:12345 weight:0 prio:0 txts:1
00:01:41.690 [I] SrpClient-----: State Updated -> ToUpdate
Done
> srp client service remove testinst _test._udp
00:01:42.136 [I] SrpClient-----: Service ToAdd -> ToRemove, "testinst" "_test._udp"
Done

00:01:42.346 [I] Settings------: Read SrpEcdsaKey
00:01:42.353 [I] SrpClient-----: Send update
00:01:42.353 [I] SrpClient-----: Service ToRemove -> Removing, "testinst" "_test._udp"
00:01:42.353 [I] SrpClient-----: State ToUpdate -> Updating
00:01:42.353 [I] SrpServer-----: Received DNS update from fdde:ad00:beef:0:d3da:5959:1956:5c20
00:01:42.363 [I] SrpServer-----: Processed DNS update info
00:01:42.363 [I] SrpServer-----:     Host:myhost.default.service.arpa.
00:01:42.363 [I] SrpServer-----:     Lease:7200, key-lease:1209600, ttl:7200
00:01:42.363 [I] SrpServer-----:     1 host address(es):
00:01:42.363 [I] SrpServer-----:       fdde:ad00:beef:0:d3da:5959:1956:5c20
00:01:42.363 [I] SrpServer-----:     Deleting service 'testinst._test._udp.default.service.arpa.'
00:01:42.363 [I] SrpServer-----: Update host myhost.default.service.arpa.
00:01:42.363 [I] SrpServer-----: Remove but retain name of service 'testinst._test._udp.default.service.arpa.'
00:01:42.363 [I] SrpServer-----: Send success response with granted lease: 7200 and key lease: 680400
00:01:42.363 [I] SrpClient-----: Received response
00:01:42.363 [I] SrpClient-----: Service Removing -> Removed, "testinst" "_test._udp"
00:01:42.363 [I] SrpClient-----: State Updating -> Updated
00:01:47.443 [I] Mle-----------: Send Advertisement (ff02:0:0:0:0:0:0:1)
00:01:47.446 [I] MeshForwarder-: Sent IPv6 UDP msg, len:90, chksum:7eb1, ecn:no, to:0xffff, sec:no, prio:net
00:01:47.446 [I] MeshForwarder-:     src:[fe80:0:0:0:ac96:56e2:3da4:719c]:19788
00:01:47.446 [I] MeshForwarder-:     dst:[ff02:0:0:0:0:0:0:1]:19788

> srp server service
testinst._test._udp.default.service.arpa.
    deleted: true
Done
```

Damian-Nordic, 2022-10-18:
@abtink Thanks a lot for the fast response :) PR looks good to me.

closed 2022-10-18 by jwhui