openthread/openthread #8311: why no data in networkdiagnostic result?

lancersky, 2022-10-19:
**Describe the bug** A clear and concise description of what the bug is.
Occasionally, there is no data in networkdiagnostic result after otbr-agent crash and restart.
[image]
[image]

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id: https://github.com/openthread/ot-br-posix/commit/3a98779dc9f0875af9c7b137e764c9663bfdc9c3
2. IEEE 802.15.4 hardware platform: EFR32MG21F768 base on GSDK v4.1.1
3. Build steps
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

jwhui, 2022-10-19:
How quickly after restart is the network diagnostic command being issued? It may be that it takes some time for the mesh topology to reform.

Are you able to provide any more visibility with logs and/or packet capture?

lancersky, 2022-10-26:
Hi @jwhui 
[syslog.txt](https://github.com/openthread/openthread/files/9865736/syslog.txt)
[networkdiagnostic-bf24792ae8781ddac84b74a55480c627.zip](https://github.com/openthread/openthread/files/9865738/networkdiagnostic-bf24792ae8781ddac84b74a55480c627.zip)

lancersky, 2022-10-26:
the normal log:
```
Wed Oct 26 02:45:56 2022 user.info otbr-agent[12656]: 21:08:37.046 [I] Platform------: Session socket is ready
Wed Oct 26 02:45:56 2022 user.info otbr-agent[12656]: 21:08:37.046 [I] Platform------: > networkdiagnostic get ff03::2 0 1 2
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.046 [D] Cli-----------: Input: networkdiagnostic get ff03::2 0 1 2
Wed Oct 26 02:45:56 2022 user.info otbr-agent[12656]: 21:08:37.047 [I] NetDiag-------: Sent diagnostic get
Wed Oct 26 02:45:56 2022 user.info otbr-agent[12656]: 21:08:37.047 [I] NetDiag-------: Received diagnostic get query
Wed Oct 26 02:45:56 2022 user.info otbr-agent[12656]: 21:08:37.047 [I] NetDiag-------: Type 0
Wed Oct 26 02:45:56 2022 user.info otbr-agent[12656]: 21:08:37.047 [I] NetDiag-------: Type 1
Wed Oct 26 02:45:56 2022 user.info otbr-agent[12656]: 21:08:37.047 [I] NetDiag-------: Type 2
Wed Oct 26 02:45:56 2022 user.info otbr-agent[12656]: 21:08:37.047 [I] NetDiag-------: Sent diagnostic get answer
Wed Oct 26 02:45:56 2022 user.info otbr-agent[12656]: 21:08:37.048 [I] NetDiag-------: Diagnostic get answer received
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.048 [D] Cli-----------: Output: DIAG_GET.rsp/ans from fdba:4397:f21e:4d9a:0:ff:fe00:9000: 00089aaa97b80efebad70102900002010f
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.048 [D] Cli-----------: Output: Ext Address: '9aaa97b80efebad7
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.048 [D] Cli-----------: Output: Rloc16: 0x9000
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.049 [D] Cli-----------: Output: Mode:
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.049 [D] Cli-----------: Output:     RxOnWhenIdle: 1
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.049 [D] Cli-----------: Output:     DeviceType: 1
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.049 [D] Cli-----------: Output:     NetworkData: 1
Wed Oct 26 02:45:56 2022 user.info otbr-agent[12656]: 21:08:37.049 [I] NetDiag-------: Sent diagnostic answer acknowledgment
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.049 [D] Mac-----------: Request to start operation "TransmitDataDirect"
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.050 [D] Mac-----------: Starting operation "TransmitDataDirect"
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.050 [D] SubMac--------: RadioState: Receive -> CsmaBackoff
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.050 [D] SubMac--------: RadioState: CsmaBackoff -> Transmit
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.050 [D] Platform------: Spinel send: cmd:PROP_VALUE_SET: prop:STREAM_RAW, iid:00, tid:03
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.061 [D] Platform------: Spinel resp: cmd:PROP_VALUE_IS, prop:LAST_STATUS, iid:00, tid:03
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.061 [D] SubMac--------: RadioState: Transmit -> Receive
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.061 [D] Mac-----------: ==============================[TX len=054]==============================
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.061 [D] Mac-----------: | 49 98 B2 08 17 FF FF 00 | 90 0D 9E 1C 00 00 01 7E | I.2............~
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.061 [D] Mac-----------: | 7A 03 00 00 02 E1 04 6D | 02 00 78 F3 FF E5 20 52 | z....a.m..xs.e R
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.062 [D] Mac-----------: | 02 AC 30 F2 0A B1 64 02 | 64 71 FF 12 03 00 01 02 | .,0r.1d.dq......
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.063 [D] Mac-----------: | 4C DE 0D AF 61 93 .. .. | .. .. .. .. .. .. .. .. | L^./a...........
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.063 [D] Mac-----------: ------------------------------------------------------------------------
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.063 [D] Mac-----------: Finishing operation "TransmitDataDirect"
Wed Oct 26 02:45:56 2022 user.info otbr-agent[12656]: 21:08:37.063 [I] MeshForwarder-: Sent IPv6 HopOpts msg, len:73, chksum:0000, ecn:no, to:0xffff, sec:yes, prio:normal
Wed Oct 26 02:45:56 2022 user.info otbr-agent[12656]: 21:08:37.064 [I] MeshForwarder-:     src:[fdba:4397:f21e:4d9a:0:ff:fe00:9000]
Wed Oct 26 02:45:56 2022 user.info otbr-agent[12656]: 21:08:37.064 [I] MeshForwarder-:     dst:[ff03:0:0:0:0:0:0:2]
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.064 [D] Mac-----------: Idle mode: Radio receiving on channel 18
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.094 [D] Mac-----------: Request to start operation "TransmitDataDirect"
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.094 [D] Mac-----------: Starting operation "TransmitDataDirect"
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.094 [D] SubMac--------: RadioState: Receive -> CsmaBackoff
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.094 [D] SubMac--------: RadioState: CsmaBackoff -> Transmit
Wed Oct 26 02:45:56 2022 user.debug otbr-agent[12656]: 21:08:37.094 [D] Platform------: Spinel send: cmd:PROP_VALUE_SET: prop:STREAM_RAW, iid:00, tid:04
[... cut ...]
Wed Oct 26 02:46:09 2022 user.info otbr-agent[12656]: 21:08:49.881 [I] MeshForwarder-: Sent IPv6 UDP msg, len:90, chksum:fce1, ecn:no, to:0xffff, sec:no, prio:net
Wed Oct 26 02:46:09 2022 user.info otbr-agent[12656]: 21:08:49.883 [I] MeshForwarder-:     src:[fe80:0:0:0:98aa:97b8:efe:bad7]:19788
Wed Oct 26 02:46:09 2022 user.info otbr-agent[12656]: 21:08:49.885 [I] MeshForwarder-:     dst:[ff02:0:0:0:0:0:0:1]:19788
Wed Oct 26 02:46:09 2022 user.debug otbr-agent[12656]: 21:08:49.886 [D] Mac-----------: Idle mode: Radio receiving on channel 18
```

jwhui, 2023-03-13:
@lancersky , are you still observing this issue?

lancersky, 2023-03-14:
hi @jwhui 
we have upgraded to GSDK4.2.1, and have not yet reproduced this issue. we will be still observing the issue. Thanks

jwhui, 2023-03-14:
@lancersky , thanks for reporting back. I'll close this issue for now, but feel free to reopen if you see it again.

closed 2023-03-14 by jwhui