openthread/openthread #8325: Low multicast delivery rates in big matter networks

markus-becker-tridonic-com, 2022-10-24:
**Describe the bug**

We have been testing multicast reliability in a network of approx. 50 Nordic DK with the matter lighting-app and matter's group functionality.

In this network we see an considerably higher loss rate of the multicast packets in REEDs compared to routers.

**To Reproduce**

1. Git commit id
NCS 2.0.2
commit aacd9eb0ba0d45483613370d8b7924fe3fa71606 (HEAD, manifest-rev)
Merge: e299df7fc f28a3e5ec
Author: Andrzej Głąbek <andrzej.glabek@nordicsemi.no>
Date:   Mon Jul 25 15:56:27 2022 +0200

2. IEEE 802.15.4 hardware platform
nRF52840 DK

3. Build steps
Binary build as distributed by NCS2.0.2.

4. Network topology
Dense topology: All devices on a table. 

**Expected behavior** 
Similar loss rates for REEDs and Routers.

**Console/log output**
N/A

**Additional context**
On request of @LuDuda we have changed `OPENTHREAD_CONFIG_MLE_CHILD_ROUTER_LINKS` to 12 and built OpenThread from sources, which resulted in a considerably better loss rate.

jwhui, 2022-11-01:
@markus-becker-tridonic-com , thanks for raising this.

I agree that increasing `OPENTHREAD_CONFIG_MLE_CHILD_ROUTER_LINKS` will improve multicast performance, as it provides more opportunities for a child to receive a given multicast message.

I think the cost of increasing `OPENTHREAD_CONFIG_MLE_CHILD_ROUTER_LINKS` is that more messages are exchanged when a child attaches to a Thread network. I recall there was a time where we did not have such a limit (the effective limit being 31 given the maximum number of active routers) and that was causing far too much MLE traffic, especially during network formation.

It may be that 12 is an appropriate compromise between something too small (e.g. 3) and something too large (e.g. 31).

@LuDuda , do you have any thoughts?

jwhui, 2022-11-04:
> Would it be possible to allow the user of the stack to configure this via the API? There is probably no one-size-fits-all setting for all sorts of scenarios.

The build time option `OPENTHREAD_CONFIG_MLE_CHILD_ROUTER_LINKS` already exists, so each product can already determine on their own what value to use. Is it necessary to provide an API to allow dynamic, runtime configuration?

> Have you tested multicast in big networks already? What was your experience with it?

It really depends on the use case. If the multicast message is large and requires fragmentation, then it is probably important to establish more router links. If the network is sparse, then there will only be a limited number of neighboring routers to work with. If the device does not require any multicast, then minimizing the number of router links helps reduce overall channel utilization.

markus-becker-tridonic-com, 2022-11-04:
Based on your second part of your answer, I would argue that an API would make sense.

Also, from an end-device manufacturer point of view, this would determine whether we could do CbS or would have to do certification by testing.

EskoDijk, 2022-12-02:
From prior experience (non-Thread 6lowpan mesh networks) I recall that sending (MPL/Trickle) multicast for lighting use cases works well with single-frame IPv6 packets, but not well for multi-frame IPv6 packets. So the best performance boost one can do is to limit the multicast packet size (preferably to only 1 frame). But ok, that's not helping you here in boosting the performance for REEDs specifically ;)

markus-becker-tridonic-com, 2023-02-08:
Closing since #8410 has been merged.

PR #8410 (2023-02-01): [mle] add API for changing number of child router links
Multicast has been proven to work unreliably with a low value of OPENTHREAD_CONFIG_MLE_CHILD_ROUTER_LINKS for big dense networks. Since there is no silver bullet value that suits all installations, make it possibly to configure its value.

Fixes #8325.
closed 2023-02-08 by markus-becker-tridonic-com