openthread/openthread #8329: Provides the KeepAlive feature for TCP

dsyx, 2022-10-26:
OpenThread supports TCP, which is generally easier to develop applications than UDP.

In a traditional home network, nodes provide some services through TCP, and usually only accept one connection per service (allowing less resource consumption). However, a client may hold a connection to a service for a long time, but not use the service. This behavior of the client may be intentional or unintentional (sending a FIN but not reaching the service node). This will make the service on the node unavailable for a long time. Supporting the TCP KeepAlive feature can avoid this trouble to a certain extent.

samkumar, 2022-10-26:
I believe that the current TCP implementation already supports keepalives. To save memory, the keepalive parameters are not tunable on a per-connection basis. All connections use the FreeBSD default parameters (see https://github.com/openthread/openthread/blob/main/third_party/tcplp/bsdtcp/tcp_timer.h#L99-L102).

These parameters requires a TCP connection to remain idle for 2 hours before any keepalive probes are sent, consistent with the FreeBSD man page for TCP: https://www.freebsd.org/cgi/man.cgi?query=tcp. I am open to making these global parameters tunable when building OpenThread.

dsyx, 2022-10-26:
You're right. It would be better if these parameters were configurable at build time.

closed 2026-11-14 by jwhui