openthread/openthread #8331: otbr-agent caught signal 11 (Segmentation fault) when Thread Certification Test(Version 1.1) 5.6.9(Router Behavior - External Route)

jinpeng1989, 2022-10-26:
otbr-agent caught signal 11 (Segmentation fault) when Thread Certification Test(Version 1.1) 5.6.9(Router Behavior - External Route).

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id: 3a98779dc9f0875af9c7b137e764c9663bfdc9c3 ot-br-posix/2022-06-02-3a98779-2; POSIX; Feb 15 2021 15:22:37
2. IEEE 802.15.4 hardware platform: SL-OPENTHREAD/2.1.0.0_GitHub-8f92d2dc8_v1; EFR32; Jul 19 2022 14:18:51
1. Git commit id
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology
[image]

**Console/log output** If applicable, add console/log output to help explain your problem.
[ThreadConsoleLog_2022_10_26_15_12.log](https://github.com/openthread/openthread/files/9869997/ThreadConsoleLog_2022_10_26_15_12.log)
[ThreadDebugLogger_2022_10_26.txt](https://github.com/openthread/openthread/files/9869999/ThreadDebugLogger_2022_10_26.txt)
[syslog.log](https://github.com/openthread/openthread/files/9870004/syslog.log)
[Test-Not-Completed-5_6_9_Router_v57.0_2022_10_26-15_13_54_Ch-23.pcapng.zip](https://github.com/openthread/openthread/files/9870007/Test-Not-Completed-5_6_9_Router_v57.0_2022_10_26-15_13_54_Ch-23.pcapng.zip)

jwhui, 2022-10-26:
@jinpeng1989 , thanks for raising this issue. Does the seg fault occur every time or intermittently?

Also, have you tried using a more recent version of OTBR?

jinpeng1989, 2022-10-26:
> @jinpeng1989 , thanks for raising this issue. Does the seg fault occur every time or intermittently?
> 
> Also, have you tried using a more recent version of OTBR?

It occur every time. Try to use [GSDK](https://github.com/SiliconLabs/gecko_sdk) from SiliconLabs, it also occur.

jwhui, 2022-10-26:
> It occur every time. Try to use [GSDK](https://github.com/SiliconLabs/gecko_sdk) from SiliconLabs, it also occur.

Can you try using a more recent version of OTBR?

suveshpratapa, 2022-10-26:
@jinpeng1989 Couple of questions.

1. I see you are using SiLabs GSDK 4.1.0.0.  Have you tried this with our two recent patch releases after this one? i.e., GSDK 4.1.1.0 or GSDK 4.1.2.0?  GSDK 4.1.2.0 will force a more recent OTBR commit (`ot-br-posix@181335224`, `openthread@2ce3d3bf0`)

2. How exactly did you set up your OTBR?  Did you change any of the default configs while installing? (params to `./script/setup` or changes to CMake / configs)

(It's curious that the seg-fault seems to happen when adding a new SRP service @jwhui )

jinpeng1989, 2022-10-26:
Ok. I'll try the new version tomorrow morning.

jinpeng1989, 2022-10-27:
The Segmentation fault can be reproduced in GSDK v4.1.3
[otbr-agent_GSDK_v4.1.3.log](https://github.com/openthread/openthread/files/9875559/otbr-agent_GSDK_v4.1.3.log)

suveshpratapa, 2022-10-27:
@jinpeng1989

The latest log shows the segmentation fault while the device is going through some kind of detach / re-attach.  Your original log showed it happening while adding an SRP service.  So seems like something happens when the network data is updated or network state (or active datasets) are changing.

- Is it possible for you to debug this issue and get us a stacktrace using gdb on your POSIX host?  You can do:
```
1. Stop the running otbr-agent service:
sudo service otbr-agent stop

2. Start agent process in gdb:
sudo gdb /usr/sbin/otbr-agent

3. Start the process in gdb context:
run -I wpan0 -B eth0 spinel+hdlc+uart:///dev/ttyACM0 trel://eth0
(or whatever your arguments to the agent are, in your /etc/default/otbr-agent file)
```

- Is there a particular set of steps that lets you reproduce this issue easily?  CLI, etc would be helpful.  We can also try to reproduce it on our end.

suveshpratapa, 2022-10-27:
@jinpeng1989 

Also going back to an earlier question:  Please let us know how you built your OTBR.  Did you modify any configs while running `./script/setup` or in your CMake environment?

jinpeng1989, 2022-10-28:
[gdb_log.log](https://github.com/openthread/openthread/files/9887594/gdb_log.log)

lancersky, 2022-10-28:
Hi,
this is the root cause.
[image]

suveshpratapa, 2022-10-28:
@lancersky @jinpeng1989 @jwhui 

👍🏼 Yup, noticed the same thing, and have a PR open here to address this issue https://github.com/openthread/openthread/pull/8338

Not sure why we haven't run into this before.

jwhui, 2022-10-29:
Fixed by #8338

closed 2022-10-29 by jwhui