openthread/openthread #8336: [tcp] Connection triggers HardFault

dsyx, 2022-10-28:
**Describe the bug**

Refer to [OpenThread CLI - TCP Example](https://github.com/openthread/openthread/blob/main/src/cli/README_TCP.md).

Made Node A perform TCP listening, and then used a Python script on the PC to test the TCP connection. The Python script performs the following actions in a loop and records the connection count and connection timeout count:

1. Connect to Node A;
2. Sleep for 50 ms;
3. Close the connection;
4. Sleep for 1 second.

After a certain number of connections (about 8000 times), Node A triggers HardFault:

```
PC = 00000610, CycleCnt = 819F8993
R0 = 7FFFFFF5, R1 = 20000060, R2 = 00000009, R3 = 00000001
R4 = 80000000, R5 = 20000060, R6 = 04404400, R7 = 7FFFFFF5
R8 = 20000060, R9 = 20000150, R10= 00000010, R11= 20000150
R12= A5A5A5A5
SP(R13)= 1FFFFFD8, MSP= 1FFFFFD8, PSP= 20001F38, R14(LR) = FFFFFFF1
XPSR = 81000003: APSR = Nzcvq, EPSR = 01000000, IPSR = 003 (HardFault)
CFBP = 00000000, CONTROL = 00, FAULTMASK = 00, BASEPRI = 00, PRIMASK = 00

FPS0 = 00000000, FPS1 = 00000000, FPS2 = 00000000, FPS3 = 00000000
FPS4 = 00000000, FPS5 = 00000000, FPS6 = 00000000, FPS7 = 00000000
FPS8 = 00000000, FPS9 = 00000000, FPS10= 00000000, FPS11= 00000000
FPS12= 00000000, FPS13= 00000000, FPS14= 00000000, FPS15= 00000000
FPS16= 00000000, FPS17= 00000000, FPS18= 00000000, FPS19= 00000000
FPS20= 00000000, FPS21= 00000000, FPS22= 00000000, FPS23= 00000000
FPS24= 00000000, FPS25= 00000000, FPS26= 00000000, FPS27= 00000000
FPS28= 00000000, FPS29= 00000000, FPS30= 00000000, FPS31= 00000000
FPSCR= 00000000
```

**To Reproduce**

1. Git commit id `2ce3d3bf0`
2. IEEE 802.15.4 hardware platform `efr32`
3. Gecko SDK Suite: Amazon 202012.00, Bluetooth 4.1.0, Bluetooth Mesh 3.0.1, EmberZNet 7.1.1.0, Flex 3.4.1.0, MCU 6.3.1.0, Micrium OS Kernel 5.13.10, OpenThread 2.1.1.0 (GitHub-2ce3d3bf0), Platform 4.1.0.0, USB 1.0.0.0, Wi-SUN 1.3.1.0, Z-Wave SDK 7.18.1.0

Node A uses the ot-cli-ftd example from Simplicity Studio (similar to [ot-efr32](https://github.com/openthread/ot-efr32)), but with a bugfix #8291.

In addition, I also modified the callback in `cli_tcp.cpp`, so that Node A will send the FIN to the peer after receiving the FIN without manually typing `tcp sendend`:

```cpp
void TcpExample::HandleTcpReceiveAvailable(otTcpEndpoint *aEndpoint,
                                           size_t         aBytesAvailable,
                                           bool           aEndOfStream,
                                           size_t         aBytesRemaining)
{
    OT_UNUSED_VARIABLE(aBytesRemaining);
    OT_ASSERT(aEndpoint == &mEndpoint);

    if (aBytesAvailable > 0)
    {
        const otLinkedBuffer *data;
        size_t                totalReceived = 0;

        IgnoreError(otTcpReceiveByReference(aEndpoint, &data));
        for (; data != nullptr; data = data->mNext)
        {
            OutputLine("TCP: Received %u bytes: %.*s", static_cast<unsigned int>(data->mLength), data->mLength,
                       reinterpret_cast<const char *>(data->mData));
            totalReceived += data->mLength;
        }
        OT_ASSERT(aBytesAvailable == totalReceived);
        IgnoreReturnValue(otTcpCommitReceive(aEndpoint, totalReceived, 0));
    }

    if (aEndOfStream)
    {
        OutputLine("TCP: Reached end of stream");
        otTcpSendEndOfStream(&mEndpoint);
    }
}

otTcpIncomingConnectionAction TcpExample::HandleTcpAcceptReady(otTcpListener *   aListener,
                                                               const otSockAddr *aPeer,
                                                               otTcpEndpoint **  aAcceptInto)
{
    OT_UNUSED_VARIABLE(aListener);

    if (mEndpointConnected)
    {
        OutputFormat("TCP: Ignoring incoming connection request from ");
        OutputSockAddr(*aPeer);
        OutputLine(" (active socket is busy)");

        return OT_TCP_INCOMING_CONNECTION_ACTION_REFUSE;
    }

    *aAcceptInto = &mEndpoint;
    return OT_TCP_INCOMING_CONNECTION_ACTION_ACCEPT;
}
```

dsyx, 2022-11-01:
I have resolved the issue, which was caused by main stack overflow. I have increased the main stack size from 1K to 2K in the ot-cli-ftd example and HardFault no longer occurs.

closed 2022-11-01 by dsyx

jwhui, 2022-11-01:
@dsyx , glad you found the issue. Thanks for reporting back!