openthread/openthread #8362: Conflicting diagnostics commands for GPIO

maciejbaczmanski, 2022-11-04:
After merge of https://github.com/openthread/openthread/pull/8316 "gpio" diag command has been added. 
The same diag command is implemented on ot-nrf528xx platform:
https://github.com/openthread/ot-nrf528xx/blob/main/src/DIAG.md#diag-gpio
Commands are overlapping now, so any attempt to use platform's specific command raises `kErrorInvalidArgs` error:
[image]

@zhanglongxia  Should one of commands' name be changed to avoid conflicts?

zhanglongxia, 2022-11-07:
I will rename the command `gpio` to `pin`.

maciejbaczmanski, 2022-11-09:
@zhanglongxia Seems good. Thanks.

zhanglongxia, 2022-11-23:
@maciejbaczmanski The OpenThread diag commands already support setting the GPIO's direction. I think you can update the Nordic's repo to support the new GPIO diag commands now.

maciejbaczmanski, 2022-12-05:
Fixed by https://github.com/openthread/openthread/pull/8391
Updated ot-nrf528xx platform implementation: https://github.com/openthread/ot-nrf528xx/pull/456

closed 2022-12-05 by maciejbaczmanski