openthread/openthread #8378: [tcplp] code scanning alert - multiplication result converted to larger type

jwhui, 2022-11-08:
Tracking issue for:
- [ ] https://github.com/openthread/openthread/security/code-scanning/1
- [ ] https://github.com/openthread/openthread/security/code-scanning/2

@samkumar, thoughts on these alerts?

samkumar, 2022-11-24:
@jwhui I'm trying to follow those links but they aren't working for me (I get GitHub's 404 page). Perhaps I need to be granted access?

jwhui, 2022-11-24:
> @jwhui I'm trying to follow those links but they aren't working for me (I get GitHub's 404 page). Perhaps I need to be granted access?

@samkumar , I adjusted the permissions. Can you try again? Thanks.

samkumar, 2022-11-24:
This is interesting --- this code is taken directly from FreeBSD. I think there are two options here.
1. I can add in casts to make the errors go away.
2. I can leave the code as is since it's taken from FreeBSD (and presumably suppress the warnings).
There seems to be no downside to adding the casts, so I lean toward Option 1. Does that sound OK to you @jwhui ?

jwhui, 2022-11-24:
@samkumar , your proposed solution (option 1) sounds good to me. Thanks.

samkumar, 2022-12-22:
I opened Pull Request #8578, which I believe will fix these code scanning alerts.

PR #8578 (2022-12-23): [tcplp] fix code scanning alerts relating to integer overflow in multipication
Fixes #8378.

I believe that the added casts will fix the code scanning alerts, but if it is possible to re-run the code scanning on this pull request to double-check, that would be great.
closed 2022-12-23 by jwhui