openthread/openthread #8400: Ubuntu 22.10 fails to create wpan interface

canisLupus1313, 2022-11-16:
After compiling and running `ot-daemon` on Ubuntu 22.10 network interface disappears and ot-daemon fails to terminate(both with ^C and `kill`).

1. Git commit id d7de66671ae8385e820f6b8405fb9527778bbe39
2. IEEE 802.15.4 hardware platform nRF52 howerer issue is on x86 host site.
3. Build steps `./script/cmake-build posix -DOT_DAEMON=ON`
4. Network topology: single board

OS logs:

```
lis 16 08:00:55 prbi-Latt-5420 ./build/posix/src/posix/ot-daemon[19173]: Running OPENTHREAD/thread-reference-20200818-2112-g3e138d3f5; POSIX; Nov 16 2022 07:59:42
lis 16 08:00:55 prbi-Latt-5420 ./build/posix/src/posix/ot-daemon[19173]: Thread version: 4
lis 16 08:00:55 prbi-Latt-5420 NetworkManager[1565]: <info>  [1668582055.7622] manager: (wpan0): new Tun device (/org/freedesktop/NetworkManager/Devices/8)
lis 16 08:00:55 prbi-Latt-5420 cinnamon[3957]: JS LOG: Unknown network device type, is 16
lis 16 08:00:55 prbi-Latt-5420 ./build/posix/src/posix/ot-daemon[19173]: Thread interface: wpan0
lis 16 08:00:55 prbi-Latt-5420 ./build/posix/src/posix/ot-daemon[19173]: RCP version: OPENTHREAD/thread-reference-20200818-2090-gd7de66671; Zephyr; Nov 15 2022 11:49:25
lis 16 08:00:57 prbi-Latt-5420 cinnamon[3957]: JS LOG: Removing a network device that was not added
lis 16 08:00:58 prbi-Latt-5420 kernel: wpan0 (unregistering): mctp_unregister: BUG mctp_ptr set for unknown type 65535
lis 16 08:00:59 prbi-Latt-5420 kernel: wpan0 (unregistered): mctp_unregister: BUG mctp_ptr set for unknown type 65535
lis 16 08:01:00 prbi-Latt-5420 kernel: wpan0 (unregistered): mctp_unregister: BUG mctp_ptr set for unknown type 65535
lis 16 08:01:01 prbi-Latt-5420 kernel: wpan0 (unregistered): mctp_unregister: BUG mctp_ptr set for unknown type 65535
lis 16 08:01:03 prbi-Latt-5420 kernel: wpan0 (unregistered): mctp_unregister: BUG mctp_ptr set for unknown type 65535
lis 16 08:01:04 prbi-Latt-5420 kernel: wpan0 (unregistered): mctp_unregister: BUG mctp_ptr set for unknown type 65535
lis 16 08:01:05 prbi-Latt-5420 kernel: wpan0 (unregistered): mctp_unregister: BUG mctp_ptr set for unknown type 65535
lis 16 08:01:06 prbi-Latt-5420 kernel: wpan0 (unregistered): mctp_unregister: BUG mctp_ptr set for unknown type 65535
lis 16 08:01:07 prbi-Latt-5420 kernel: wpan0 (unregistered): mctp_unregister: BUG mctp_ptr set for unknown type 65535
lis 16 08:01:07 prbi-Latt-5420 kernel: unregister_netdevice: waiting for wpan0 to become free. Usage count = 2
```

Kernel version:

```
(pbida@prbi-Latt-5420)(✗)(10:26:40)
(openthread)$uname -a
Linux prbi-Latt-5420 5.19.0-23-generic #24-Ubuntu SMP PREEMPT_DYNAMIC Fri Oct 14 15:39:57 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

wgtdkp, 2022-11-16:
The `wpan0` is already in use, try specify a different network interface name such as `wpan1`

canisLupus1313, 2022-11-16:
@wgtdkp this is happening at the first run. Changing the interface name does not help.

closed 2022-12-05 by canisLupus1313

nickgarvey, 2022-12-13:
I had this issue when I clicked the "FORM" button in the UI when there already was a network. Just not clicking that button worked for me and didn't show the kernel errors. Ubuntu 21.10

agners, 2022-12-13:
I've encountering this too trying to run otbr-agent on ArchLinux. I run it from within a separate network namespace to rule out any interference from other software. I am starting the OTBR without a useful radio, and stop it using Ctrl+C, and then see this errors:

```
wpan0 (unregistering): mctp_unregister: BUG mctp_ptr set for unknown type 65535
wpan0 (unregistered): mctp_unregister: BUG mctp_ptr set for unknown type 65535
```

To me this seems like a kernel bug?

@mkj I see in the kernel log that you recently changed the error message there, do you maybe have thoughts on this?

jk-ozlabs, 2022-12-14:
Ah, looking at #8523, is the type being changed there? we'd probably need to handle that case in the MCTP driver...

agners, 2022-12-14:
Yes the link type gets changed via `TUNSETLINK` to `ARPHRD_VOID`.

I don't see a good reason why it gets changed, so I suggest to explicitly set the current default in https://github.com/openthread/openthread/pull/8526.

But I guess ideally this should still be correctly handled in the MCTP driver.

jk-ozlabs, 2022-12-14:
> But I guess ideally this should still be correctly handled in the MCTP driver.

Yep, agreed- @mkj has just posted a fix to netdev:

https://patchwork.kernel.org/project/netdevbpf/patch/20221214061044.892446-1-matt@codeconstruct.com.au/

canisLupus1313, 2022-12-14:
@agners @jk-ozlabs #8526 solves the issue thanks.