openthread/openthread #8429: SRP service adding failed - error:Security

lancersky, 2022-11-22:
**Describe the bug** A clear and concise description of what the bug is.
I use Raspberry Pi as OTBR for testing.
On SRP server side
```
iot@raspberrypi:~/openthread/ot-br-posix$ sudo ot-ctl dataset init new
Done
iot@raspberrypi:~/openthread/ot-br-posix$ sudo ot-ctl dataset commit active
Done
iot@raspberrypi:~/openthread/ot-br-posix$ sudo ot-ctl ifconfig up
Done
iot@raspberrypi:~/openthread/ot-br-posix$ sudo ot-ctl thread start
Done
iot@raspberrypi:~/openthread/ot-br-posix$ sudo ot-ctl state 
leader
Done
iot@raspberrypi:~/openthread/ot-br-posix$ sudo ot-ctl dataset active
Active Timestamp: 0
Channel: 11
Channel Mask: 0x07fff800
Ext PAN ID: dead00beef00cafe
Mesh Local Prefix: fdde:ad00:beef:0::/64
Network Key: 7bf66ea7252ec281d90ed9f242aa7d2c
Network Name: OpenThread
PAN ID: 0x8b4d
PSKc: d862280c866494839ff54427701be8d9
Security Policy: 672 onrc
Done
iot@raspberrypi:~/openthread/ot-br-posix$ sudo ot-ctl dataset active -x
0e080000000000000000000300000b35060004001fffe00208dead00beef00cafe0708fddead00beef000005107bf66ea7252ec281d90ed9f242aa7d2c030a4f70656e54687265616401028b4d0410d862280c866494839ff54427701be8d90c0402a0f7f8
Done
iot@raspberrypi:~/openthread/ot-br-posix$  sudo service otbr-agent status
● otbr-agent.service - OpenThread Border Router Agent
     Loaded: loaded (/lib/systemd/system/otbr-agent.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2022-11-22 06:53:51 UTC; 9min ago
    Process: 687 ExecStartPre=service mdns start (code=exited, status=0/SUCCESS)
   Main PID: 971 (otbr-agent)
      Tasks: 1 (limit: 4416)
     Memory: 4.6M
        CPU: 944ms
     CGroup: /system.slice/otbr-agent.service
             └─971 /usr/sbin/otbr-agent -I wpan0 -B wlan0 spinel+hdlc+uart:///dev/ttyACM0 trel://wlan0

Nov 22 07:02:54 raspberrypi otbr-agent[971]: 00:08:50.927 [I] MeshForwarder-:     dst:[fdde:ad00:beef:0:a8c8:a39:a1e0:7de4]:53535
Nov 22 07:02:54 raspberrypi otbr-agent[971]: 00:08:50.927 [I] SrpServer-----: Received DNS update from fdde:ad00:beef:0:4bdd:1c37:f2aa:ae03
Nov 22 07:02:54 raspberrypi otbr-agent[971]: 00:08:50.958 [I] SrpServer-----: SRP update handler is notified (updatedId = 1917516210)
Nov 22 07:02:54 raspberrypi otbr-agent[971]: [INFO]-ADPROXY-: Advertise SRP service updates: host=dev4312.default.service.arpa.
Nov 22 07:02:54 raspberrypi otbr-agent[971]: [INFO]-ADPROXY-: Failed to advertise SRP service updates (id = 1917516210)
Nov 22 07:02:54 raspberrypi otbr-agent[971]: 00:08:50.958 [I] SrpServer-----: Handler result of SRP update (id = 1917516210) is received: InvalidArgs
Nov 22 07:02:54 raspberrypi otbr-agent[971]: 00:08:50.958 [W] SrpServer-----: Send fail response: 5
Nov 22 07:02:54 raspberrypi otbr-agent[971]: 00:08:50.963 [I] MeshForwarder-: Sent IPv6 UDP msg, len:60, chksum:20c2, ecn:no, to:0x1001, sec:yes, prio:low
Nov 22 07:02:54 raspberrypi otbr-agent[971]: 00:08:50.963 [I] MeshForwarder-:     src:[fdde:ad00:beef:0:a8c8:a39:a1e0:7de4]:53535
Nov 22 07:02:54 raspberrypi otbr-agent[971]: 00:08:50.963 [I] MeshForwarder-:     dst:[fdde:ad00:beef:0:4bdd:1c37:f2aa:ae03]:49154
iot@raspberrypi:~/openthread/ot-br-posix$ sudo ot-ctl srp server state
running
Done
```
On SRP client side,
```
> factoryreset
> 
> dataset set active 0e080000000000000000000300000b35060004001fffe00208dead00beef00cafe0708fddead00beef000005107bf66ea7252ec281d90ed9f242aa7d2c030a4f70656e54687265616401028b4d0410d862280c866494839ff54427701be8d90c0402a0f7f8
Done
> ifconfig up
Done
> thread start
Done
> state
child
Done
> ipaddr
fd84:5ca7:ee4c:1:298d:7536:e305:79a6
fdde:ad00:beef:0:0:ff:fe00:1001
fdde:ad00:beef:0:4bdd:1c37:f2aa:ae03
fe80:0:0:0:40d8:3b26:c7de:e954
Done
> srp client callback enable
Done
> srp client autostart enable
Done
>  srp client host name dev4312
Done
> srp client host address fdde:ad00:beef:0:4bdd:1c37:f2aa:ae03
Done
> srp client service add ins1 _udp.tcp 12345
Done
> SRP client callback - error:Security
Host info:
    name:"dev4312", state:ToAdd, addrs:[fdde:ad00:beef:0:4bdd:1c37:f2aa:ae03]
Service list:
    instance:"ins1", name:"_udp.tcp", state:Adding, port:12345, priority:0, weight:0
SRP client callback - error:Security
Host info:
    name:"dev4312", state:ToAdd, addrs:[fdde:ad00:beef:0:4bdd:1c37:f2aa:ae03]
Service list:
    instance:"ins1", name:"_udp.tcp", state:Adding, port:12345, priority:0, weight:0
SRP client callback - error:Security
Host info:
    name:"dev4312", state:ToAdd, addrs:[fdde:ad00:beef:0:4bdd:1c37:f2aa:ae03]
Service list:
    instance:"ins1", name:"_udp.tcp", state:Adding, port:12345, priority:0, weight:0
SRP client callback - error:Security
Host info:
    name:"dev4312", state:ToAdd, addrs:[fdde:ad00:beef:0:4bdd:1c37:f2aa:ae03]
Service list:
    instance:"ins1", name:"_udp.tcp", state:Adding, port:12345, priority:0, weight:0
SRP client callback - error:Security
Host info:
    name:"dev4312", state:ToAdd, addrs:[fdde:ad00:beef:0:4bdd:1c37:f2aa:ae03]
Service list:
    instance:"ins1", name:"_udp.tcp", state:Adding, port:12345, priority:0, weight:0
SRP client callback - error:Security
Host info:
    name:"dev4312", state:ToAdd, addrs:[fdde:ad00:beef:0:4bdd:1c37:f2aa:ae03]
Service list:
    instance:"ins1", name:"_udp.tcp", state:Adding, port:12345, priority:0, weight:0
SRP client callback - error:Security
Host info:
    name:"dev4312", state:ToAdd, addrs:[fdde:ad00:beef:0:4bdd:1c37:f2aa:ae03]
Service list:
    instance:"ins1", name:"_udp.tcp", state:Adding, port:12345, priority:0, weight:0
SRP client callback - error:Security
```

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id: https://github.com/openthread/ot-br-posix/commit/1813352247aa60fb8993773918f1e5b4af6f3b79
2. IEEE 802.15.4 hardware platform: nrf52840 USB dongle
3. Build steps: `./script/bootstrap && INFRA_IF_NAME=wlan0 ./script/setup`
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

superwhd, 2022-11-22:
I think the format of your service type is wrong. 
```
srp client service add ins1 _udp.tcp 12345
```

Here you need to have a underscore `_` before the transport protocol `tcp`.

Could you try following?
```
srp client service add ins1 _udp._tcp 12345
```

lancersky, 2022-11-22:
You are right. And I need to clear the error service instance before adding a new one. Thanks!

closed 2022-11-22 by lancersky