openthread/openthread #8441: ff03::02 scope broadcast storm

jinpeng1989, 2022-11-24:
There are 1 Leader and 15 Router in thread network. The ff03::02 destination address packets be retransmission back and forth indefinitely. 
What are the conditions that a Router repeat packet whose destination address is ff03::02? And what are the conditions that the Router drop the packet? Has the Router remember the packets been repeat?
[image]
[image]

jinpeng1989, 2022-11-24:
Add wireshark packets. networkkey: fdbbf7fc9a27c8839ae75390b803ea51
[broadcast_storm_00000_20221124142138.zip](https://github.com/openthread/openthread/files/10083769/broadcast_storm_00000_20221124142138.zip)
[broadcast_storm_00001_20221124145427.zip](https://github.com/openthread/openthread/files/10083785/broadcast_storm_00001_20221124145427.zip)
[broadcast_storm_00002_20221124152138.zip](https://github.com/openthread/openthread/files/10083795/broadcast_storm_00002_20221124152138.zip)
[broadcast_storm_00003_20221124154546.zip](https://github.com/openthread/openthread/files/10083797/broadcast_storm_00003_20221124154546.zip)
[broadcast_storm_00004_20221124161745.zip](https://github.com/openthread/openthread/files/10083798/broadcast_storm_00004_20221124161745.zip)

jwhui, 2022-11-28:
The multicast packets that I see are primarily Address Query (`/a/aq`) or Diagnostics Query (`/d/dq`) packets. Address Query messages are sent whenever the device is trying to send a unicast packet to the Target IPv6 Address contained within the Address Query payload. Diagnostics Query packets are sent whenever the application requests to send a Diagnostics Query.

Note that the IPv6 multicast messages are not all the same. The IPv6 multicast packets contain an MPL Option which has a Sequence field. The different Sequence values indicate that these are actually different packets.

jinpeng1989, 2022-11-29:
Could you show me where is the MPL Option?
[image]

jinpeng1989, 2022-11-29:
All the Sequence in MPL Option are the same.
[image]

jinpeng1989, 2022-11-29:
Is it expected that a packet destined for address ff03::02 be forwarded indefinitely? With the same token and message ID.

jwhui, 2022-11-30:
Can you provide more information about your setup? For example:
1. Git commit id
2. IEEE 802.15.4 hardware platform
3. Build configs that you are using

Also, can you provide more information so that others can reproduce this issue? For example, what application-layer traffic is being generated?

jinpeng1989, 2022-11-30:
There are 1 Border Router and 15 REED on the thread network. 
Border Router:
1. Git commit id  ot-br-posix GSDK v4.1.3
2. IEEE 802.15.4 hardware platform :  EFR32MG21F768 base on GSDK v4.1.3

The information about REED will append later.

jinpeng1989, 2022-11-30:
This issue was discovered by accident, and we don't know the recurrence rule. 
Our Border Router server a Thread Network topologies web page,  it will call 'ubus call otbr networkdata' interface periodically(per 10 seconds) when this page is opened.

jinpeng1989, 2022-12-06:
The multicast packet is discarded if the IPv6 Hop Limit value becomes zero. The IPv6 Hop Limit value on the current version otbr-agent we use is 64. I think the value is too large. 
[image]
The define on Thread v1.3.0-RC1 Specification_clean.pdf is 36.
[image]
A Thread Network contains a maximum of 32 routers, so I think 31 is enough.

jinpeng1989, 2022-12-06:
As specification define, MAY send an initial multicast with an IPv6 Hop Limit of 1 or 2. If this initial multicast fails to elicit a response, a full Realm-Local multicast is sent. Whether the protocol stack currently supports this mechanism? In our test environment, all the devices were within each other's radio range. I think the IPv6 Limit of 1 or 2 is enough.
[image]

jinpeng1989, 2022-12-06:
If the IPv6 Hop Limit value becomes zero, the multicast packet will be discarded. Whether there are other circumstances will stop forwarding before Hop Limit value becomes zero?

On the second Thread network where there was no multicast storm. The network also have 16 Routers. The Address Query packet been forwarded 24 times. The Hop Limit value in the source packet is 64. The Hop Limit value in all the other packets is 63. I have two questions:
1 Why can the device stop forwarding? What condition?
2 The IPv6 Hop Limit field of an MPL Data Message is decreased by one by an MPL Forwarder,
7 prior to starting the forwarding process for this message. There are 16 Routers in this network and been forwarded 24 times. It seems that some router do not decrease the Hop Limit?
[image]

jwhui, 2022-12-06:
Thread uses MPL to forward IPv6 multicast messages ([RFC 7731](https://www.rfc-editor.org/rfc/rfc7731)). Thread Routers maintain a cache of recently seen MPL Sequence values for a given MPL Seed to avoid forwarding the message again. Thread Routers should be relying on the MPL cache to determine whether or not to forward the IPv6 multicast message. It should not be relying on the IPv6 Hop Limit value.

One possibility could be that Thread Routers are not retaining the MPL cache entry long enough. We just merged #8482 as one potential fix for this. You can see if it helps your situation.

jinpeng1989, 2022-12-06:
Renew the lifetime of the Seed set entry when receive the same packet. I think it's going to improve a lot. 

The OPENTHREAD_CONFIG_MPL_SEED_SET_ENTRY_LIFETIME is 5 second. The delta time between two forward packet may be longer than 5S. So I don't think it's going to solve every situation. We will apply this fix and test it.
[image]

jwhui, 2023-01-26:
Closing stale issue.

closed 2023-01-26 by jwhui