openthread/openthread #8443: Posix: add option to stop setting ML-EID as deprecated when border routing is disabled

HiFiPhile, 2022-11-24:
**Is your feature request related to a problem? Please describe.** A clear and concise description of what the problem is.
As discussed in #8200, Mesh Local addresses are set to `deprecated` which makes the kernel not to choose those as source address for traffics initiated from linux host.

While it's an intended behavior of border router, it's not convenient in case of border router is disabled, as the source address need always be specified.

In our case we are building an IoT controller who has Internet access but with an isolated Thread network, so I set `-DOT_BORDER_ROUTER=OFF` and adding a default route will affect Internet access.

**Describe the solution you'd like** A clear and concise description of what you want to happen.
When border routing is disabled, stop setting ML-EID address as deprecated.

**Describe alternatives you've considered** A clear and concise description of any alternative solutions or features you've considered.
Always set ML-EID address while needed, but in production environment it's hard to do:
- Too many work to add source address option for all applications
- Can't identify ML-EID from other addresses (RLOC, ALOC)

**Additional context** Add any other context or screenshots about the feature request here.
[image]

MaikVermeulen, 2025-05-02:
This seems to affect our application as well. Any workarounds at the moment?

jwhui, 2025-05-22:
Is it possible for the service running on the BR to explicitly bind to the ML-EID address, explicitly indicating what IPv6 source address to use?