openthread/openthread #8461: [Thread 1.3.x] related OT PRs and Thread JIRAs.

abtink, 2022-11-30:
Created this issue to track OT PRs and related Thread Technical Committee JIRAs.

- [ ] [SPEC-1046](https://threadgroup.atlassian.net/browse/SPEC-1046) Child supervision 
  - https://github.com/openthread/openthread/pull/7993
- [ ] [SPEC-1061](https://threadgroup.atlassian.net/browse/SPEC-1061) delay TLV for qry

_Note: Access to the Thread JIRA links requires Thread membership and Thread group login/account._

This is follow-up to https://github.com/openthread/openthread/issues/7460.

suveshpratapa, 2023-09-19:
@abtink Does this represent the latest remaining work for 1.3.1 on GitHub? Or are there any items missing?

abtink, 2023-09-19:
> Does this represent the latest remaining work for 1.3.1 on GitHub? Or are there any items missing?

That's the intention behind this.

suveshpratapa, 2023-09-19:
Thank you @abtink ! I was going to ask you about the network diagnostics work item so thank you for pointing me to that. Also I know the TCAT/BBTC PR is currently open and yet to be merged so will that count as another 1.3.x item to be added to this list?

abtink, 2023-09-19:
This tracks everything for 1.3.x (not just 1.3.1). I created and use this issue to track stuff that I did or plan to work on myself (to have a place with all the links to SPEC JIRAs and PRs). 

Feel free to create a new issues to track other items if you want (or you can add them here)

BTW, SRP replication is not part of 1.3.1 and DNS Push is not supported in OT(can be supported by platform if desired).

EskoDijk, 2024-09-11:
@abtink Seeing that all issues we wanted to do, that were listed here, are addressed, we can probably close this issue? See #9930 for the open issue on remaining items for 1.4.

abtink, 2024-09-11:
SGTM.

closed 2024-09-11 by abtink