openthread/openthread #8519: The Child failed send MAC Data Frames to its Parent when its Parent send announce message to another channel.

jinpeng1989, 2022-12-12:
The Child failed send MAC Data Frames to its Parent when its Parent leaves the current channel  to send announce message to another channel.
The rloc16 of this Child is 0x2805， it consecutive unicast MAC Data Frames sent to 0x2800， but there is no Mac layer Ack.
[image]
Checked the log, the Parent is sending announce on other channels at this time. What is the maximum time when Parent leaves the current channel to send announce message to another channel? Is there a definition in the standard? 
Is this kind of packet loss as expected?
[image]

jinpeng1989, 2022-12-12:
Add log and wireshark packets, Network Key: 5d7838199c00a8dd596dd687ecb1b84e
[5d7838199c00a8dd596dd687ecb1b84e.zip](https://github.com/openthread/openthread/files/10205225/5d7838199c00a8dd596dd687ecb1b84e.zip)
[send_failed_when_announce.log](https://github.com/openthread/openthread/files/10205226/send_failed_when_announce.log)

jwhui, 2022-12-13:
The MLE Announce message should take on the order of tens of milliseconds - basically the time it takes for the radio to switch to a different channel, transmit a broadcast message, and return back to the main channel.

It appears that your device is only attempting to transmit a unicast message a total of 4 times (unless the ack message is getting dropped from the sniffer capture). Thread Specification requires 15 retransmissions when no ack is received and OpenThread implements that requirement.

jinpeng1989, 2022-12-13:
The sleep end device in our system base on the Thread Specification 1.1.1, the macMaxFrameRetries is 3.
[image]

jwhui, 2022-12-13:
In OpenThread, the number of retries is controled via the build config [`OPENTHREAD_CONFIG_MAC_DEFAULT_MAX_FRAME_RETRIES_DIRECT`](https://github.com/openthread/openthread/blob/b02bdb9e3bfc0207737bebb94824713f61f5352c/src/core/config/mac.h#L73). I would encourage you to increase the number of retries to 15. Even better if you can upgrade your device to Thread 1.3, which includes many other enhancements such as this.

jwhui, 2023-01-26:
Closing stale issue.

closed 2023-01-26 by jwhui