openthread/openthread #8522: otbr-agent crash during/right-after "NetDataPublshr: DNS/SRP service (state:Adding)"

AlanLCollins, 2022-12-13:
**Describe the bug** The otbr-agent crash with SIGSEV right-after "NetDataPublshr: DNS/SRP service (state:Adding)" gets printed. I can't reproduce constantly but another hint I found is that only happens after the "Slaac---------: Adding address" & when NetDataPublshr finds "preferred:1" (It does not happen with preferred:0).
`12-13 10:35:00.602  5248  5248 I otbr-agent: [I] Settings------: Read SlaacIidSecretKey
12-13 10:35:00.602  5248  5248 I otbr-agent: [I] Slaac---------: Adding address fde0:3d41:cdd5:1:5baf:941f:6bb1:9e71
12-13 10:35:00.602  5248  5248 I otbr-agent: [I] Platform------: [netif] Sent request#4 to add fde0:3d41:cdd5:1:5baf:941f:6bb1:9e71/64
12-13 10:35:00.603  5248  5248 I otbr-agent: [I] BorderRouter--: Start evaluating routing policy, scheduled in 3227 milliseconds
12-13 10:35:00.603  5248  5248 I otbr-agent: [I] NetDataPublshr: DNS/SRP service (state:Adding) in netdata - total:1, preferred:1, desired:2`

FULL LOGS here: 
[16.02.22.22_G2A1EF052126043J_logcat.txt](https://github.com/openthread/openthread/files/10221506/16.02.22.22_G2A1EF052126043J_logcat.txt)

I am still investigating. I will update this ticket when I find new information.

jwhui, 2022-12-14:
@AlanLCollins , thanks for reporting this issue.

Can you indicate the git commit you are using? We've recently fixed some bugs related to format strings when logging. One example:
- https://github.com/openthread/openthread/pull/8338

AlanLCollins, 2022-12-15:
@jwhui 
I got good initial results. Our QA team is running stress & longevity tests to confirm the crashes are gone. I will keep you posted on the results. Thanks !

jwhui, 2022-12-15:
> @jwhui I got good initial results. Our QA team is running stress & longevity tests to confirm the crashes are gone. I will keep you posted on the results. Thanks !

Great, thanks for the update!

closed 2022-12-20 by AlanLCollins

jwhui, 2022-12-20:
> @jwhui the issue got resolved by cherry-picking your suggestions. Thank you !

@AlanLCollins , great to hear that fixed your issue. Thanks for reporting back!