openthread/openthread #8527: The otbr-agent does not respond to ot-ctl command and restful request(localhost:8081/diagnostics), it seems that the main thread of otbr-agent entered an endless loop.

jinpeng1989, 2022-12-14:
The otbr-agent does not respond to ot-ctl command and restful request(localhost:8081/diagnostics), can respond for ubus. It seems that the main thread of otbr-agent entered an endless loop. 

The uptime of this OTBR device is 2 day, the problem appeared after the device had been running for a day, do not recover now. 

Build steps
There are 100 devices in this system, 1 OTBR + 15 REED + 85 SED.
Network topology
[image]
syslog for otbr-agent

The issue is difficult to reproduce. What other information can help locate the issue? 
[syslog_otbr-agent.log](https://github.com/openthread/openthread/files/10223937/syslog_otbr-agent.log)

jinpeng1989, 2022-12-14:
Is there a mechanism to monitor threads entering an endless loop?

jwhui, 2022-12-14:
This appears to be a duplicate of #8510 .

jwhui, 2023-02-13:
Closing stale issue.

closed 2023-02-13 by jwhui