openthread/openthread #8533: Incorrect use of kParentRequestReedTimeout instead of kMaxChildUpdateResponseTimeout in mle.cpp

ADEscobar, 2022-12-15:
Hello,

I think in mle.cpp the delay after a Child ID request is incorrectly set to kParentRequestReedTimeout:

    if (HasAcceptableParentCandidate() && (SendChildIdRequest() == kErrorNone))
    {
        SetAttachState(kAttachStateChildIdRequest);
        delay = kParentRequestReedTimeout;
        ExitNow();
    }

It should use kMaxChildUpdateResponseTimeout instead. In fact that variable is not used anywhere else, so I think it should belong there an be like this:

    if (HasAcceptableParentCandidate() && (SendChildIdRequest() == kErrorNone))
    {
        SetAttachState(kAttachStateChildIdRequest);
        delay = kMaxChildUpdateResponseTimeout;
        ExitNow();
    }

Best,
Antonio

abtink, 2022-12-15:
Thanks @ADEscobar,

If I recall correctly, the exact wait time (for child to receive "Child ID Response") is not explicitly specified in the Thread sepc, and the OT implementation basically reused the constant `kParentRequestReedTimeout` here (as a long enough wait time).

Not sure about using `kMaxChildUpdateResponseTimeout` for this as this is about "Child ID Response" and not the "Child Update Response" message. 

I traced the definition of `kMaxChildUpdateResponseTimeout` constant back to PR #992, and it was used in this method to set as "timeout" on the `Child` entries in child table that are being restored (after router is rebooted and tries to recover its previous entries (which are restored from non-volatile settings).
```c++
void MleRouter::HandleChildUpdateRequestTimer(void)
{
    VerifyOrExit(GetDeviceState() == kDeviceStateRouter || GetDeviceState() == kDeviceStateLeader, ;);

    for (int i = 0; i < mMaxChildrenAllowed; i++)
    {
        if (mChildren[i].mState == Neighbor::kStateRestored)
        {
            SendChildUpdateRequest(&mChildren[i]);
            mChildren[i].mState = Neighbor::kStateChildUpdateRequest;

            if ((mChildren[i].mMode & ModeTlv::kModeRxOnWhenIdle) != 0)
            {
                mChildren[i].mTimeout = Timer::MsecToSec(kMaxChildUpdateResponseTimeout);
            }

            mChildUpdateRequestTimer.Start(kChildUpdateRequestPeriod);
            break;
        }
    }

exit:
    {}
}
```
The `main` code today uses the saved `timeout` value as originally specified by the child (in `ChildTable::Restore()`) so the `kMaxChildUpdateResponseTimeout` is now unused (which I think would be good to remove).

ADEscobar, 2022-12-15:
Thanks, I see! 

Then I think a new one (kMaxChildIdResponseTimeout) should be added instead of reusing kParentRequestReedTimeout, since we already have a kMaxChildIdRequestTimeout, it would be much clearer. 

The parent request timeouts (kParentRequestRouterTimeout and kParentRequestReedTimeout) actually allow the child to wait for a while for parent responses, so I think it is also useful to decouple that.

abtink, 2022-12-15:
I agree adding new constant would be good -> Submitted https://github.com/openthread/openthread/pull/8536

ADEscobar, 2022-12-16:
Should this one also be renamed without the max?

kMaxChildIdRequestTimeout

Since it is not a random delay I think it would be better as kChildIdRequestTimeout.

abtink, 2022-12-16:
In the PR we add `kChildIdResponseTimeout` as a new constant and `kMaxChildUpdateResponseTimeout` is removed.

abtink, 2022-12-16:
I see, you are talk about another existing constant. We can rename it.

I am sure there are other possibilities to rename constants and methods to make things more clear (I always prefer longer and more descriptive names that read like english... but key is not to overdo it) 😄

ADEscobar, 2022-12-16:
Yes, just because max and timeout, at least in this context, is  a bit of an oxymoron (max wait maybe?), and also to follow the nomenclature you introduced without "max", but of course it is not very important. :)

PR #8536 (2022-12-19): [mle] define constant for "Child ID Response" timeout
This commit defines `kChildIdResponseTimeout` as a new constant in `mle_types.hpp` which is used as maximum wait time to receive a "Child ID Response" message from parent during attach. This commit also removes the now unused `kMaxChildUpdateResponseTimeout`.

---

 Resolves #8533
closed 2022-12-19 by jwhui