openthread/openthread #8557: If a Child that has attached to the Parent indicating it is an rx-on-when-idle Device try to use Child Update Request to modify its mode to rx-off-when-idle, should the parent reject it?

jinpeng1989, 2022-12-20:
As stated in the Specification that A Child that has attached to the Parent indicating it is an rx-on-when-idle Device MUST NOT use Child Update Request to modify its mode to rx-off-when-idle. During the actual testing, there is a Child attempt to do this.
[image]

Packet 98: The Child Device initially attached to the Parent indicating it is an rx-on-when-idle Device.
Packet 128: The Child Device use Child Update Request to modify its mode to rx-off-when-idle. 
Packet 140: The Parent Device reply Child Update Respond without 'error'. 

Is there a consequence to that? Will it affect the rest of the communication? If the answer is yes, I think the Parent should reject the attempting from the Child Device.
[image]
[image]
[image]

More details, please see attachment, Network Key: f7f05cdd3cda3c5d3c983d2f597b177e
[rxon_to_rxoff.zip](https://github.com/openthread/openthread/files/10265424/rxon_to_rxoff.zip)

jinpeng1989, 2022-12-20:
This change applies to the Child side, right?

jwhui, 2022-12-20:
> This change applies to the Child side, right?

Yes, correct.

jinpeng1989, 2022-12-20:
The Child Device cannot be updated to the new version of the stack due to memory resource constraints. I wonder if this fix can be applied to the lower version stack?
Is it possible that the Parent reject the Child's attempting? Then the Child will try to re-attach.

jwhui, 2023-01-26:
Closing stale issue.

closed 2023-01-26 by jwhui