openthread/openthread #8604: The Link Metrics Status Sub-TLV in Link Metrics Management Response message is 254(Failure - Other failure)

jinpeng1989, 2023-01-03:
The Test 7 1 1 Single Shot With Enhanced Acks failed during the Thread Certification test. Is there a build config for this feature. Or are we using too low a version of the stack?

Border Router:
Git commit id ot-br-posix GSDK v4.1.3
IEEE 802.15.4 hardware platform : EFR32MG21F768 base on GSDK v4.1.3

[image]

[image]

Network Key: 33221100445566778899aabbccddeeff
[Fail_L_7_1_1_Leader_v57.0_2022_12_26-17_47_06_Ch-11.zip](https://github.com/openthread/openthread/files/10334604/Fail_L_7_1_1_Leader_v57.0_2022_12_26-17_47_06_Ch-11.zip)
[Fail_L_7_1_1_Leader_v57.0_2022_12_26-17_50_36.pdf](https://github.com/openthread/openthread/files/10334605/Fail_L_7_1_1_Leader_v57.0_2022_12_26-17_50_36.pdf)

jinpeng1989, 2023-01-09:
Any updates on this issue?

Irving-cl, 2023-01-09:
Hi @jinpeng1989,

Could you check if you have enabled the option `OPENTHREAD_CONFIG_MLE_LINK_METRICS_SUBJECT_ENABLE` when building RCP? And please also make sure if `otPlatRadioConfigureEnhAckProbing` have been implemented on EFR platform. For example:
* On simulation platform: https://github.com/openthread/openthread/blob/main/examples/platforms/simulation/radio.c#L1345
* On Nordic platform: https://github.com/openthread/ot-nrf528xx/blob/main/src/src/radio.c#L1412

Let's first check if these things are good. Then we can try to add some logs in [LinkMetrics::ConfigureEnhAckProbing](https://github.com/openthread/openthread/blob/main/src/core/thread/link_metrics.cpp#L616) and see which steps did it return the error.

BTW, I think we'd better discuss certification related issue on JIRA, instead of github issues. So please create JIRAs for certification related issues. I would love to help.

jwhui, 2023-02-08:
Closing stale issue.

closed 2023-02-08 by jwhui