openthread/openthread #8613: multicast routing failed using openwrt

seba0506, 2023-01-04:
Dear Jonathan Hui,

I have enabled the below-mentioned feature in my application. After enabling the "OTBR_BACKBONE_ROUTER=ON" feautre otbr-agent was stopped.

**Makefile**
-------------

#
#  Copyright (c) 2020, The OpenThread Authors.
#  All rights reserved.
#
#  Redistribution and use in source and binary forms, with or without
#  modification, are permitted provided that the following conditions are met:
#  1. Redistributions of source code must retain the above copyright
#     notice, this list of conditions and the following disclaimer.
#  2. Redistributions in binary form must reproduce the above copyright
#     notice, this list of conditions and the following disclaimer in the
#     documentation and/or other materials provided with the distribution.
#  3. Neither the name of the copyright holder nor the
#     names of its contributors may be used to endorse or promote products
#     derived from this software without specific prior written permission.
#
#  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
#  AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
#  IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
#  ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE
#  LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
#  CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF
#  SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS
#  INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN
#  CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
#  ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
#  POSSIBILITY OF SUCH DAMAGE.
#

LOCAL_SOURCE_DIR:=$(abspath $(realpath $(dir $(lastword $(MAKEFILE_LIST))))/../../..)

include $(TOPDIR)/rules.mk

PKG_NAME:=openthread-br
PKG_LICENSE:=BSD-3-Clause
PKG_LICENSE_FILES:=LICENSE
PKG_RELEASE:=1
PKG_VERSION:=1.0

#PKG_SOURCE_PROTO:=git
#PKG_SOURCE_URL:=https://github.com/openthread/ot-br-posix.git
#PKG_MIRROR_HASH:=skip

#PKG_SOURCE:=$(PKG_NAME)-$(PKG_SOURCE_DATE).tar.gz
#PKG_SOURCE_SUBDIR:=$(PKG_NAME)
#PKG_BUILD_DIR:=$(BUILD_DIR)/$(PKG_NAME)

include $(INCLUDE_DIR)/package.mk
include $(INCLUDE_DIR)/cmake.mk

CMAKE_SOURCE_DIR=$(LOCAL_SOURCE_DIR)
CMAKE_OPTIONS+= \
        -DBUILD_TESTING=OFF \
        -DCMAKE_INSTALL_PREFIX=/usr \
        -DOTBR_OPENWRT=ON \
        -DOTBR_BORDER_AGENT=ON \
        -DOTBR_BORDER_ROUTING=ON \
        -DOTBR_SRP_ADVERTISING_PROXY=ON \
        -DOTBR_MDNS=avahi \
        -DOTBR_INFRA_IF_NAME=br-lan \
        -DOTBR_POSIX_SETTINGS_PATH=\"/etc/openthread\" \
        -DOTBR_READLINE=OFF \
        -DOTBR_FIREWALL=ON \
        -DOTBR_BACKBONE_ROUTER=ON \
[... 342 lines cut ...]
udp        0      0 127.0.0.1:58000         0.0.0.0:*                           11115/eapd
udp        0      0 127.0.0.1:58032         0.0.0.0:*                           11247/wps_pbcd
udp        0      0 127.0.0.1:46000         0.0.0.0:*                           11115/eapd
udp        0      0 127.0.0.1:45000         0.0.0.0:*                           11115/eapd
udp        0      0 127.0.0.1:44000         0.0.0.0:*                           11115/eapd
udp        0      0 127.0.0.1:45032         0.0.0.0:*                           11128/bsd
udp        0      0 0.0.0.0:5353            0.0.0.0:*                           7225/avahi-daemon:
udp        0      0 127.0.0.1:43000         0.0.0.0:*                           11115/eapd
udp        0      0 127.0.0.1:44032         0.0.0.0:*                           2733/wlssk
udp        0      0 127.0.0.1:40200         0.0.0.0:*                           8672/mesh_broker
udp        0      0 127.0.0.1:42000         0.0.0.0:*                           11115/eapd
udp        0      0 127.0.0.1:52000         0.0.0.0:*                           11115/eapd
udp        0      0 127.0.0.1:42032         0.0.0.0:*                           11231/acsd2
udp        0      0 127.0.0.1:51000         0.0.0.0:*                           11115/eapd
udp        0      0 127.0.0.1:52032         0.0.0.0:*                           11117/ceventd
udp        0      0 0.0.0.0:68              0.0.0.0:*                           12528/odhcpc
udp        0      0 127.0.0.1:50000         0.0.0.0:*                           11115/eapd
udp        0      0 :::123                  :::*                                8002/ntpd
udp        0      0 :::40888                :::*                                7225/avahi-daemon:
udp        0      0 :::5353                 :::*                                7225/avahi-daemon:
udp        0      0 :::546                  :::*                                12660/odhcp6c
udp        0      0 :::547                  :::*                                6866/odhcpd
root@CP2235RA916:~#

Did I miss anything? please suggest

seba0506, 2023-01-04:
Dear Jonathan Hui,

I have changed the below-mentioned code, and after that backbone, and router working fine without any issues. I have used the SO_REUSEADDR flag in setscokopt api. Can you please confirm from your end?

**Modified code**
---------------------

void MulticastRoutingManager::InitMulticastRouterSock(void)
{
    int                 one = 1;
    struct icmp6_filter filter;
    struct mif6ctl      mif6ctl;

    // Create a Multicast Routing socket
    mMulticastRouterSock = SocketWithCloseExec(AF_INET6, SOCK_RAW, IPPROTO_ICMPV6, kSocketBlock);
    VerifyOrDie(mMulticastRouterSock != -1, OT_EXIT_ERROR_ERRNO);

    // Enable Multicast Forwarding in Kernel
    //VerifyOrDie(0 == setsockopt(mMulticastRouterSock, IPPROTO_IPV6, MRT6_INIT, &one, sizeof(one)), OT_EXIT_ERROR_ERRNO);
    VerifyOrDie(0 == setsockopt(mMulticastRouterSock, IPPROTO_IPV6, SO_REUSEADDR, &one, sizeof(one)), OT_EXIT_ERROR_ERRNO);

    // Filter all ICMPv6 messages
    ICMP6_FILTER_SETBLOCKALL(&filter);
    VerifyOrDie(0 == setsockopt(mMulticastRouterSock, IPPROTO_ICMPV6, ICMP6_FILTER, (void *)&filter, sizeof(filter)),
                OT_EXIT_ERROR_ERRNO);

    memset(&mif6ctl, 0, sizeof(mif6ctl));
    mif6ctl.mif6c_flags     = 0;
    mif6ctl.vifc_threshold  = 1;
    mif6ctl.vifc_rate_limit = 0;

    // Add Thread network interface to MIF
    mif6ctl.mif6c_mifi = kMifIndexThread;
    mif6ctl.mif6c_pifi = if_nametoindex(gNetifName);
    VerifyOrDie(mif6ctl.mif6c_pifi > 0, OT_EXIT_ERROR_ERRNO);
    //VerifyOrDie(0 == setsockopt(mMulticastRouterSock, IPPROTO_IPV6, MRT6_ADD_MIF, &mif6ctl, sizeof(mif6ctl)),OT_EXIT_ERROR_ERRNO);
     VerifyOrDie(0 == setsockopt(mMulticastRouterSock, IPPROTO_IPV6, SO_REUSEADDR, &mif6ctl, sizeof(mif6ctl)),OT_EXIT_ERROR_ERRNO);

    // Add Backbone network interface to MIF
    mif6ctl.mif6c_mifi = kMifIndexBackbone;
    mif6ctl.mif6c_pifi = if_nametoindex(gBackboneNetifName);
    VerifyOrDie(mif6ctl.mif6c_pifi > 0, OT_EXIT_ERROR_ERRNO);
    //VerifyOrDie(0 == setsockopt(mMulticastRouterSock, IPPROTO_IPV6, MRT6_ADD_MIF, &mif6ctl, sizeof(mif6ctl)),OT_EXIT_ERROR_ERRNO);
   VerifyOrDie(0 == setsockopt(mMulticastRouterSock, IPPROTO_IPV6, SO_REUSEADDR, &mif6ctl, sizeof(mif6ctl)),OT_EXIT_ERROR_ERRNO);
}

jwhui, 2025-01-07:
Closing stale issue.

closed 2025-01-07 by jwhui