openthread/openthread #8614: Discovering networks when acting as SED

Damian-Nordic, 2023-01-04:
**Describe the bug**
Matter over Thread device must support `ScanNetworks` command that can be issued before and after the device has joined a Thread network. That command is implemented using `otThreadDiscover` on OpenThread platforms.

I noticed `otThreadDiscover` does not work reliably when the device is attached to its parent as a SED. That is, the device detaches from its parent and struggles to discover any networks. Therefore, I came up with a solution - temporary switching to _rx-on-when-idle_ while `otThreadDiscover` is working: https://github.com/project-chip/connectedhomeip/pull/21279. It worked fine until https://github.com/openthread/openthread/commit/6f1631aeba08f8c19022f01abb9fc861fd80360a which started re-attaching to the parent when switching from _rx-on-when-idle_ to _rx-off-when-idle_. In the Matter case it causes connectivity issues when trying to send the scan results back to the controller.

As far as I understand comments in https://github.com/openthread/openthread/issues/8557, the change was meant to address the scenario in which the device first attaches as a non-SED and later switches to a SED mode (which is forbidden by the spec), so maybe the logic could be adjusted to allow for temporary switches without re-attaching?

Or maybe:
1. `otThreadDiscover` should work for a SED and we should first investigate why it doesn't?
2. `otThreadDiscover` was not designed to be called when being already attached and Matter `ScanNetworks` command should not be called after commissioning?

**To Reproduce**

I'll provide reproduction steps once we agree what is the desired path :)

**Expected behavior**

Either, `otThreadDiscover` works when being in a SED mode, or temporary transition to a non-SED mode and back works without losing the connection to the parent.

**Console/log output**

**Additional context**

Damian-Nordic, 2023-01-23:
Looks like https://github.com/openthread/openthread/pull/8498 (which I haven't noticed before) solves the problem with temporary switching to non-SED mode. Still, I'm leaving this issue open to potentially discuss the topic of MLE discovery causing SED disconnects.

AlanLCollins, 2023-01-30:
I agree with @Damian-Nordic . otThreadDiscover shall be allowed for sleepy end devices. 
The current workaround injects unnecessary Child update flow, that leads to other issues like: https://github.com/openthread/openthread/issues/8681

abtink, 2023-03-04:
I just want point out that performing "Discovery Scan" on SED will consume energy (device need to stay in RX mode for the entire scan duration).
 
I would say SEDs ideally should not scan after attaching to network or if they do, the impact on battery life should be taken into account.

LuDuda, 2023-03-31:
Closed by #8832

closed 2023-03-31 by LuDuda