openthread/openthread #8618: What causes IPv6 UDP messages to be dropped?

lancersky, 2023-01-05:
**Describe the bug** A clear and concise description of what the bug is.
Topology: OTBR <---> two End Device(nrf52840)
The end device sends a CoAP message to the OTBR, but the messages are dropped. So, I cannot receive messages in my application.
```
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.452 [D] Mac-----------: Received frame from short address 0x2002
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.452 [D] Mac-----------: Rx security - frame counter 12199
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.452 [D] Mac-----------: ==============================[RX len=122]==============================
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.452 [D] Mac-----------: | 79 98 61 90 B7 00 20 02 | 20 0D A7 2F 00 00 01 C1 | y.a.7. . .'/...A
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.453 [D] Mac-----------: | 80 78 0C 7E 55 E4 F2 86 | 08 94 9F 41 82 1D B6 B9 | .x.~Udr....A..69
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.453 [D] Mac-----------: | 29 FE 34 DA 7B F0 DB 38 | 16 33 9D 51 58 03 7C A4 | )~4Z{p[8.3.QX.|$
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.453 [D] Mac-----------: | 60 6D AA A2 66 26 0C A8 | B6 73 74 61 74 75 73 FF | `m*"f&.(6status.
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.453 [D] Mac-----------: | 7B 22 76 65 72 73 69 6F | 6E 22 3A 22 4F 50 45 4E | {"version":"OPEN
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.453 [D] Mac-----------: | 54 48 52 45 41 44 2F 30 | 37 35 39 64 65 34 66 63 | THREAD/0759de4fc
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.454 [D] Mac-----------: | 3B 20 5A 65 70 68 79 72 | 3B 20 4A 61 6E 20 20 34 | ; Zephyr; Jan  4
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.454 [D] Mac-----------: | 20 32 30 32 7E 27 F4 18 | B4 00 .. .. .. .. .. .. |  202~'t.4.......
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.454 [D] Mac-----------: ------------------------------------------------------------------------
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.461 [D] Mac-----------: Idle mode: Radio receiving on channel 26
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.462 [D] Mac-----------: Received frame from short address 0x2002
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.465 [D] Mac-----------: Rx security - frame counter 12200
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.465 [D] Mac-----------: ==============================[RX len=122]==============================
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.468 [D] Mac-----------: | 79 98 62 90 B7 00 20 02 | 20 0D A8 2F 00 00 01 E1 | y.b.7. . .(/...a
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.469 [D] Mac-----------: | 80 78 0C 0F 33 20 31 38 | 3A 33 36 3A 34 33 22 2C | .x..3 18:36:43",
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.471 [D] Mac-----------: | 22 74 68 72 65 61 64 5F | 76 65 72 73 69 6F 6E 22 | "thread_version"
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.472 [D] Mac-----------: | 3A 34 2C 22 65 75 69 36 | 34 22 3A 22 39 34 38 33 | :4,"eui64":"9483
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.473 [D] Mac-----------: | 63 34 33 61 61 39 64 39 | 62 38 39 65 22 2C 22 65 | c43aa9d9b89e","e
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.475 [D] Mac-----------: | 78 74 61 64 64 72 22 3A | 22 30 61 31 62 32 63 36 | xtaddr":"0a1b2c6
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.476 [D] Mac-----------: | 39 34 61 63 63 37 61 34 | 31 22 2C 22 61 64 64 72 | 94acc7a41","addr
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.477 [D] Mac-----------: | 22 3A 22 66 55 8B 0D 34 | B4 00 .. .. .. .. .. .. | ":"fU..44.......
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.478 [D] Mac-----------: ------------------------------------------------------------------------
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.479 [D] Mac-----------: Idle mode: Radio receiving on channel 26
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.481 [D] Mac-----------: Received frame from short address 0x2002
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.483 [D] Mac-----------: Rx security - frame counter 12201
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.485 [D] Mac-----------: ==============================[RX len=122]==============================
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.487 [D] Mac-----------: | 79 98 63 90 B7 00 20 02 | 20 0D A9 2F 00 00 01 E1 | y.c.7. . .)/...a
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.488 [D] Mac-----------: | 80 78 0C 1B 64 31 31 3A | 32 32 3A 30 3A 30 3A 63 | .x..d11:22:0:0:c
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.490 [D] Mac-----------: | 62 30 65 3A 39 66 36 65 | 3A 35 32 64 39 3A 37 61 | b0e:9f6e:52d9:7a
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.492 [D] Mac-----------: | 31 31 22 2C 22 72 6C 6F | 63 31 36 22 3A 38 31 39 | 11","rloc16":819
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.493 [D] Mac-----------: | 34 2C 22 73 77 5F 76 65 | 72 22 3A 22 30 2E 32 2E | 4,"sw_ver":"0.2.
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.495 [D] Mac-----------: | 37 22 2C 22 64 61 74 61 | 22 3A 7B 22 74 65 6D 70 | 7","data":{"temp
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.495 [D] Mac-----------: | 65 72 61 74 75 72 65 22 | 3A 32 39 2E 35 36 37 37 | erature":29.5677
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.497 [D] Mac-----------: | 39 33 39 39 19 DF 88 3C | B4 00 .. .. .. .. .. .. | 9399._.<4.......
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.499 [D] Mac-----------: ------------------------------------------------------------------------
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.500 [D] Mac-----------: Idle mode: Radio receiving on channel 26
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.501 [D] Mac-----------: Received frame from short address 0x2002
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.503 [D] Mac-----------: Rx security - frame counter 12202
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.504 [D] Mac-----------: ==============================[RX len=098]==============================
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.506 [D] Mac-----------: | 69 98 64 90 B7 00 20 02 | 20 0D AA 2F 00 00 01 E1 | i.d.7. . .*/...a
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.507 [D] Mac-----------: | 80 78 0C 27 39 39 39 39 | 39 39 39 2C 22 68 75 6D | .x.'9999999,"hum
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.508 [D] Mac-----------: | 69 64 69 74 79 22 3A 33 | 34 2E 32 37 31 32 33 39 | idity":34.271239
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.509 [D] Mac-----------: | 39 39 39 39 39 39 39 39 | 39 2C 22 6C 69 67 68 74 | 999999999,"light
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.510 [D] Mac-----------: | 22 3A 31 32 30 34 2C 22 | 70 72 65 73 73 22 3A 31 | ":1204,"press":1
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.512 [D] Mac-----------: | 30 31 2E 37 35 36 34 39 | 36 7D 7D 00 30 6C D2 62 | 01.756496}}.0lRb
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.513 [D] Mac-----------: | 21 E0 .. .. .. .. .. .. | .. .. .. .. .. .. .. .. | !`..............
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.515 [D] Mac-----------: ------------------------------------------------------------------------
Thu Jan  5 04:00:05 2023 user.info otbr-agent[7696]: 00:58:51.516 [I] MeshForwarder-: Received IPv6 UDP msg, len:384, chksum:9d51, ecn:no, from:0x2002, sec:yes, prio:normal, rss:-32.0
Thu Jan  5 04:00:05 2023 user.info otbr-agent[7696]: 00:58:51.517 [I] MeshForwarder-:     src:[fdde:ad00:beef:0:e4f2:8608:949f:4182]:56120
Thu Jan  5 04:00:05 2023 user.info otbr-agent[7696]: 00:58:51.518 [I] MeshForwarder-:     dst:[fdde:ad00:beef:0:1db6:b929:fe34:da7b]:5683
Thu Jan  5 04:00:05 2023 user.debug otbr-agent[7696]: 00:58:51.519 [D] Mac-----------: Idle mode: Radio receiving on channel 26
[... cut ...]
44970 5d fddead00beef00009a6c13ddf87d5c08d120 s 2000
Done
```
**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id **GSDK4.2.0**
2. IEEE 802.15.4 hardware platform **EFR32MG21F768/GSDK4.2.0**
3. Build steps
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

abtink, 2023-01-05:
One suggestion: Check who has the address `fdde:ad00:beef:0:1db6:b929:fe34:da7b`? 

Looks to me message is successfully received, but since not for this device it is being fowarded onwards (put in the tx queue) and then the msg tx is dropped (probably since no known destination?)

lancersky, 2023-01-05:
Thanks for your suggestion. It should be that my End Device did not update the server IP after the OTBR factory reset and re-create the same Thread network.

closed 2023-01-05 by lancersky